jneilliii/OctoPrint-BedLevelVisualizer #2: nice plugin so here is my experience with it

djunisol, 2018-04-16:
doesnt work with manual mesh bed leveling included in th3d 
think cause the answer is different to what it expects
hope it helps below is the code my printer spits out:

Send: M420 V
Recv: Mesh Bed Level data:
Recv: Num X,Y: 5,5
Recv: Z offset: 0.00000
Recv: Measured points:
Recv:         0        1        2        3        4
Recv:  0 +0.00000 -0.02500 -0.02500 +0.00000 +0.10000
Recv:  1 -0.00000 +0.02500 -0.02500 -0.02500 +0.07500
Recv:  2 -0.00000 -0.00000 +0.02500 +0.02500 +0.10000
Recv:  3 -0.00000 -0.02500 -0.02500 -0.00000 +0.05000
Recv:  4 -0.00000 -0.02500 -0.02500 -0.02500 +0.02500
Recv: 
Recv: echo:Bed Leveling Uit
Recv: echo:Fade Height Uit
Recv: ok

mudcruzr, 2018-04-16:
I would really love to see this work with bilinear levelling too, here's what mine spits out from G29 T1 onwards:

Send: G29 T1
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3
Recv:  0 -0.033 -0.139 -0.061 +0.206
Recv:  1 +0.020 -0.058 +0.042 +0.334
Recv:  2 +0.030 +0.010 +0.115 +0.424
Recv:  3 -0.028 -0.061 +0.085 +0.447
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5        6        7        8        9
Recv:  0 -0.03317 -0.07513 -0.11710 -0.13869 -0.13330 -0.10752 -0.06080 +0.01401 +0.10977 +0.20553
Recv:  1 -0.01400 -0.05327 -0.09254 -0.11143 -0.10332 -0.07483 -0.02535 +0.05232 +0.15098 +0.24964
Recv:  2 +0.00517 -0.03140 -0.06798 -0.08416 -0.07335 -0.04214 +0.01011 +0.09063 +0.19219 +0.29375
Recv:  3 +0.01960 -0.01297 -0.04554 -0.05829 -0.04508 -0.01204 +0.04221 +0.12522 +0.22944 +0.33367
Recv:  4 +0.02862 +0.00326 -0.02210 -0.02963 -0.01465 +0.01815 +0.07255 +0.15701 +0.26307 +0.36912
Recv:  5 +0.03291 +0.01607 -0.00077 -0.00236 +0.01407 +0.04575 +0.09953 +0.18508 +0.29274 +0.40039
Recv:  6 +0.02965 +0.01830 +0.00694 +0.00955 +0.02788 +0.06017 +0.11508 +0.20302 +0.31357 +0.42412
Recv:  7 +0.01541 +0.00404 -0.00734 -0.00367 +0.01739 +0.05348 +0.11266 +0.20528 +0.32098 +0.43668
Recv:  8 -0.00637 -0.02080 -0.03523 -0.03224 -0.00803 +0.03361 +0.09879 +0.19740 +0.31955 +0.44171
Recv:  9 -0.02814 -0.04564 -0.06313 -0.06080 -0.03344 +0.01374 +0.08492 +0.18952 +0.31813 +0.44673
Recv: 
Recv: X:191.00 Y:195.00 Z:15.45 E:0.00 Count X:15280 Y:15600 Z:6328
Recv: ok

jneilliii, 2018-04-16:
What firmware are each of you using?  The plugin only works if your firmware supports the Bed Topography Report in csv format.  For Marlin that gcode is G29 T1.  Other reports, like these you are showing won't know how to be interpreted without some code rewrites.

jneilliii, 2018-04-16:
I've figured out the problem and am working on a fix for non-ubl users.

mudcruzr, 2018-04-16:
Sorry, just saw your earlier comment.
I'm using Marlin bugfix-2.0.x and bilinear levelling

jneilliii, 2018-04-17:
Going to use #3 to track this issue, closing this ticket.

closed 2018-04-17 by jneilliii

jneilliii, 2018-04-17:
Added new release 0.0.3 to add non-UBL support. Give it a try and report back. I did some cursory testing with both bi-linear and UBL and it seems to work with both, make sure to select the correct "Data Collector Flag" in settings.

I specifically would like to know if the orientation of the graph display relative to the x/y axes is correct.