jneilliii/OctoPrint-BedLevelVisualizer #3: can't get past the  Please wait, retrieving current mesh.

mvl03, 2018-04-16:
My printer requires g28 prior to g29 t1 so from the get go the visualized gets stuck on Please wait, , retrieving current mesh.

I have to reboot system, restart browser and manually perform g28 prior to opening the visualizer (once opened it won't get out of the please wait state). 

After that if I open visualizer it will perform g29 but then still get stuck on please wait. Terminal does show some results 

Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 +0.002 +0.062 -0.025
Recv:  1 -0.005 +0.022 -0.028
Recv:  2 +0.005 -0.048 -0.010
Recv: 
Recv: X:262.00 Y:175.00 Z:5.81 E:0.00 Count A:34977 B:6963 Z:2316
Recv: ok

I have not seen anything past the please wait...

jneilliii, 2018-04-16:
What firmware are you using? The plugin only works if your firmware supports the Bed Topography Report in csv format. For Marlin that gcode is G29 T1. Other reports, like these you are showing won't know how to be interpreted without some code rewrites.

jneilliii, 2018-04-16:
I'm working on this, the flag being used to trigger the data collection is different based on the leveling method used.  Should have an update today to resolve this for you guys.

mvl03, 2018-04-16:
Marlin bug_fix 1.1.8 in case it helps. Thx

NickRimmer, 2018-04-16:
I've same problem with marlin

jneilliii, 2018-04-17:
Added new release 0.0.3 to add non-UBL support.  Give it a try and report back.  I did some cursory testing with both bi-linear and UBL and it seems to work with both, make sure to select the correct "Data Collector Flag" in settings.

I specifically would like to know if the orientation of the graph display relative to the x/y axes is correct.

closed 2018-04-17 by jneilliii

oliboehm, 2018-04-17:
Now it works with Marlin 1.1.8 and Bilinear Leveling. Also orientation of axis is correct (with my configuration).

It would be nice if visualizer will not start automatically when I switch to the tab. A button to start it manually gives more control.

mvl03, 2018-04-17:
Thx, that worked!!