jneilliii/OctoPrint-BedLevelVisualizer #4: Prusa i3 MK2/s MK2.5 MK3 G80 and G81 support

3d-gussner, 2018-04-16:
Hi,

nice plugin, is it hard to implement this for a Prusa i3 MK2/s, MK2.5 and MK3 as these use different gcode?
G80 meshbedleveling and G81 showing values.

jneilliii, 2018-04-16:
You may want to try this one...it's not interactive, but looks to be made for your situation.

https://github.com/ff8jake/OctoPrint-PrusaMeshMap

jneilliii, 2018-04-17:
@3d-gussner, the most recent version that I just released may work for you, but I won't know unless I have some additional information related to the output of the G81 command from you.  You can definitely give it a try.  Basically you need a flag that tells this plugin to start collecting data, so for example in the following response...

```
Send: G29 T1
Recv: echo:Home XYZ first
Recv: 
Recv: Bed Topography Report for CSV:
Recv: 
Recv: 0.256,0.170,0.140,0.535,0.183,0.159,0.054,0.197,0.105,0.105
Recv: 0.488,0.277,0.174,-0.049,0.125,0.020,0.027,0.070,-0.006,-0.006
Recv: 0.479,0.197,0.220,0.062,0.116,0.062,-0.053,0.169,0.038,0.038
Recv: 0.315,0.306,0.126,0.079,-0.096,-0.008,0.014,-0.105,0.011,0.127
Recv: 0.444,0.362,0.162,0.084,-0.160,-0.100,-0.074,-0.079,0.038,0.154
Recv: 0.506,0.293,0.195,0.274,0.093,-0.040,0.040,-0.083,-0.131,-0.131
Recv: 0.583,0.265,0.174,0.517,0.042,-0.021,-0.083,-0.094,-0.130,-0.130
Recv: 0.425,0.267,0.270,0.370,0.073,-0.125,-0.075,-0.057,-0.056,-0.056
Recv: 0.499,0.356,0.280,0.266,0.056,0.053,-0.021,0.120,-0.171,-0.171
Recv: 0.438,0.265,0.188,-0.154,0.141,0.005,-0.041,-0.117,-0.135,-0.135
Recv: ok P15 B3
```
I use the text 

`Bed Topography Report for CSV:`

for the Data Collector Flag in settings, and it will grab all the text received until it receives an OK response, and then process.

3d-gussner, 2018-04-18:
@jneilliii Prusa uses G80 for mesh bed leveling and G81 to show the values, here an example:
>>>G80
SENDING:G80
echo:Enqueing to the front: "G28 W0"
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
clean up finished
babystep applied
FrontLeft  a =17 microns.
FrontRight c =-14 microns.
MidRight   d =14 microns.
RearRight  e =12 microns.
RearLeft   g =15 microns.
MidLeft  h =13 microns.
Bed leveling correction finished
Upsample finished
Mesh bed leveling activated
echo:busy: processing
echo:busy: processing
echo:busy: processing
Go home finished
>>>G81
SENDING:G81
Num X,Y: 7,7
Z search height: 5
Measured points:
  0.18250  0.07876  -0.00652  -0.07333  -0.12169  -0.15157  -0.16300
  0.15857  0.07071  0.00579  -0.03620  -0.05526  -0.05137  -0.02456
  0.14374  0.06672  0.01609  -0.00815  -0.00599  0.02256  0.07750
  0.13800  0.06678  0.02439  0.01083  0.02611  0.07022  0.14317
  0.14135  0.07089  0.03069  0.02074  0.04105  0.09162  0.17244
  0.15380  0.07906  0.03498  0.02157  0.03883  0.08675  0.16533
  0.17533  0.09128  0.03728  0.01333  0.01944  0.05561  0.12183

And the result i get from http://lokspace.eu/3d-printer-auto-bed-leveling-mesh-visualizer/
[image]

3d-gussner, 2018-04-18:
Thanks for adding the option to define custom gcode. As the script expect "," between the values (G29 T1 vs G29) and Prusa G81 only generates "  " (2x Space) it never finish. It would be great if we could define the delimiter.
You should add a kind of timeout if the results aren't processed in time.

jneilliii, 2018-04-18:
Yep, the latest version should work for you. Use Measured points: for your flag. The one other requirement is that an ok is received after the mesh data to stop collecting data.

3d-gussner, 2018-04-18:
Even after a reboot get 'Please wait, retrieving current mesh.' and cannot see any retry in the terminal windows that G80 and G81 are used. Is there any data that has to be deleted?

jneilliii, 2018-04-18:
Send me a screenshotof your settings.

jneilliii, 2018-04-18:
Both space and comma are accounted for in the code.

3d-gussner, 2018-04-18:
Here my setting:

[image]

And that is going for ever and stays also after reboot, deinstall and reinstall of the plug-in.

[image]

And here the Terminal results:

[image]

Thanks a lot taking time

icemansid, 2018-04-18:
It seems to be ignoring the X0Y0 values. You can see from the attached photos that what it stores is not what is printed via the terminal.
[image]
[image]

jneilliii, 2018-04-18:
@3d-gussner: make your settings look like @icemansid in the above post. 

@icemansid: I think I know what the issue is, the spaces are getting doubled up. I'll do some tweaking to the collector tonight.

icemansid, 2018-04-18:
Awesome! Thanks!

3d-gussner, 2018-04-18:
@icemansid @jneilliii Thanks, didn't change the Data Collector Flag to 'Measured points:'
[image]

[image]

[image]

Great Job @jneilliii

icemansid, 2018-04-18:
@jneilliii  It is now working with appropriate values at 0/0. Thanks!

jneilliii, 2018-04-18:
I think you guys may still be experiencing the double space issue based on the stored mesh data. There should only be one tab between each number and I'm seeing a couple on that last screenshot @3d-gussner

jneilliii, 2018-04-19:
@3d-gussner, @icemansid: prior to me releasing the next version to resolve the extra tab issue I would appreciate you test it out for me, since I don't have a Prusa.  If you open OctoPrint plugin manager and use the following url to install it will update your instance.  If it doesn't prompt you to reload after restarting, please press ctrl+f5 on your keyboard to update your local browser cache with the updated code.  Once upgraded, just run through an update process and the stored mesh data will look more in-line with each other and your graph shouldn't look like a football.

3d-gussner, 2018-04-19:
@jneilliii Thanks for the update but the result is the same, the plot is identical to 0.4 with xy 0 to 6 and to results from http://lokspace.eu/3d-printer-auto-bed-leveling-mesh-visualizer/

[image]

vs.

[image]

But the 'Stored Mesh Data' is still missing the 1st column ...
[image]

vs.

[image]

jneilliii, 2018-04-19:
Thanks, that helps.  I do shift each row for the other formats, didn't notice previously that the G81 command doesn't have a row index. It will be more like the csv export.  Let me adjust the code and get you to test again.

3d-gussner, 2018-04-19:
It is a pity that Prusa firmware is so far away from Marlin as since the split/fork they didn't merge lot of things from Marlin 'home'. Thanks for all the effort you putting into it.

BTW in your v 0.5 README.md you wrote
**Fixed**
  - Remove duplicated tabs from Prusa firmware's M81 response.

it should be G81 :-)

jneilliii, 2018-04-19:
Thanks for the heads up.  I just updated the 0.0.5 branch.  Give that a try again and see if we are at least collecting all the columns into the stored mesh. I added a logger line, that will also write out to octoprint.log each line received as part of the collector.  That could also help me in debugging the duplicate tabs.  I did add in something that in theory should work to remove the duplicate tabs, but I'm not sure it's working.

3d-gussner, 2018-04-19:
@jneilliii Now it is not working anymore here a part of the octoprint.log

2018-04-19 04:40:35,826 - octoprint.plugins.bedlevelvisualizer - INFO - 0.19583  0.08681  -0.00235  -3  -0.15074  -0.16050
2018-04-19 04:40:35,839 - octoprint.util.comm - ERROR - Error while processing hook bedlevelvisualize
Traceback (most recent call last):
  File "/home/chip/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.8-py2.7.egg/octopri, line 2438, in _readline
    ret = hook(self, ret)
  File "/home/chip/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_bedlevelvisualizer/__in4, in processGCODE
    self._logger.info("converted to:" + new_line);
TypeError: cannot concatenate 'str' and 'list' objects
2018-04-19 04:40:35,883 - octoprint.plugins.bedlevelvisualizer - INFO - 0.17154  0.08010  0.01209  -0  -0.05143  -0.02576
2018-04-19 04:40:35,889 - octoprint.util.comm - ERROR - Error while processing hook bedlevelvisualize
Traceback (most recent call last):
  File "/home/chip/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.8-py2.7.egg/octopri, line 2438, in _readline
    ret = hook(self, ret)
  File "/home/chip/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_bedlevelvisualizer/__in4, in processGCODE
    self._logger.info("converted to:" + new_line);
TypeError: cannot concatenate 'str' and 'list' objects
2018-04-19 04:40:35,897 - octoprint.plugins.bedlevelvisualizer - INFO - 0.15559  0.07636  0.02338  -0  0.02201  0.07407
2018-04-19 04:40:35,900 - octoprint.util.comm - ERROR - Error while processing hook bedlevelvisualize
Traceback (most recent call last):
  File "/home/chip/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.8-py2.7.egg/octopri, line 2438, in _readline
    ret = hook(self, ret)
  File "/home/chip/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_bedlevelvisualizer/__in4, in processGCODE
    self._logger.info("converted to:" + new_line);
TypeError: cannot concatenate 'str' and 'list' objects

3d-gussner, 2018-04-19:
@jneilliii Thanks it works now. Here the results:
[image]

and Terminal:

[image]

3d-gussner, 2018-04-19:
I guess we can close the issue. Thanks again for all the effort and time.

3d-gussner, 2018-04-19:
Here a small test with 3 blades on the right side of the bed, just for fun:

[image]

3d-gussner, 2018-04-19:
Have another issue:
G80 needs low case letters like 'G80 a10 b-10 c-20 d14 e20 f45 g78 h68' to correct via gcode and not LCD but the BedLevelVirtualizer sends it in CAPS 'Send: G80 A10 B-10 C-20 D14 E20 F45 G78 H68'

jneilliii, 2018-04-19:
That issue is not my plugin, but OctoPrint in general.  You need to go into settings and adjust the "Terminal Auto Uppercase Blacklist" and add G80 to it.

[image]

jneilliii, 2018-04-19:
Closed with release of 0.0.5.

closed 2018-04-19 by jneilliii

kdorff, 2018-07-24:
Thanks! This seems to work great for me on my MK3.

timur-tabi, 2018-10-21:
I know this issue is closed, but is the resulting information documented anywhere?  I was hoping to see a file that lists various printers and how each one should be configured, but I can't find anything like that anywhere.