jneilliii/OctoPrint-BedLevelVisualizer #7: Error: Unknown Command ""

token47, 2018-04-16:
I'm having a problem, here is the log:

```
Recv: ok
[...]
Send: M155 S0
[...]
Recv: ok
Send: G28
Recv: echo:busy: processing
Printer seems to support the busy protocol, adjusting timeouts and setting busy interval accordingly
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: X:105.00 Y:118.00 Z:11.48 E:0.00 Count X:8400 Y:9440 Z:4592
Recv: ok
Send: G29 T
Recv: echo:Unknown command: ""
Recv: ok
Send: M113 S2
Recv: ok
```

My config have the following commands:

```
G28
G29 T
```
If I put just G29 there, it complains that I must home first, so I put them together. As you can see in the above log, the G28 is executed correctly (and the printer does home) and then it gives an error on the second command.

(I'm using Bilinear ABL as noted on other issue, but nevertheless it should get past this point and get stuck with the matrix report syntax instead).

jneilliii, 2018-04-17:
What do you get if you type just G29 or M420 V?  What's your firmware?

token47, 2018-04-17:
The firmware is Marlin v1.1.8, have not tried with only "G29" because without the T parameter the topology does not get printed on the log output. Will try and report back.

jneilliii, 2018-04-17:
That's the version I have running. Not quite sure why the command would not be recognized by the firmware. Double check you don't have any extra spaces between the 9 and T maybe?

jneilliii, 2018-04-19:
Curious if G29 T1 or other `T#` option would be recognized?

trux66, 2018-04-19:
Sorry to jump in but I am running the Marlin 1.1.8 firmware as well. I do get a response from the G28 and G29 T1 commands but it doesn't look like the sample. I'd also like to get the visualizer to work.

Here is an extract of what those commands return on my system.
```
Recv: FIRMWARE_NAME:Marlin 1.1.8 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:TEVO Tarantula (EasyConfig) EXTRUDER_COUNT:1
[...]
Send: G28
[...]
Recv: X:100.00 Y:12.00 Z:5.50 E:0.00 Count X:8000 Y:960 Z:2200
Recv: ok
[...]
Send: G29 T1
[...]
Recv: 
Recv: Bed Height Topography:
Recv:    +--- BACK --+
Recv:    |           |
Recv:  L |    (+)    | R
Recv:  E |           | I
Recv:  F | (-) N (+) | G
Recv:  T |           | H
Recv:    |    (-)    | T
Recv:    |           |
Recv:    O-- FRONT --+
Recv:  (0,0)
Recv:  +0.72083 -0.29567 -0.62317
Recv:  +0.65783 +0.06883 -0.64917
Recv:  +0.77283 +0.12333 -0.77567
Recv: 
Recv: X:189.95 Y:101.99 Z:6.27 E:0.00 Count X:15200 Y:8160 Z:2200
Recv: ok
[...]
```

jneilliii, 2018-04-19:
You could also use `(0,0)`, as that is closer to the mesh data and would reduce additional  line processing, not that you'll probably notice a difference.

token47, 2018-04-20:
Sorry, have been away from the printer, will test it asap.

BTW, there is no T1 in Bilinear, only T (no number). The T# is only supported in UBL.

jneilliii, 2018-04-20:
About to release a new version that will fix some issues reported with view bindings.

jneilliii, 2018-04-20:
Version 0.0.6 just released.  Give it a try and let me know how it goes.  If you inadvertently got this version already during my development, you will have to reinstall manually.

jslay88, 2018-04-20:
0.0.6 is working for me now. I had to (I think) save twice for it to store the `Data Collector Flag`, once I had it where it persisted, it finally spit out a lovely image.

This is my Marlin 1.1.8 output, which is now parsing correctly.

```
Send: G29
Recv: State: Off
Recv: Num X,Y: 3,3
Recv: Z offset: 0.00000
Recv: Measured points:
Recv:         0        1        2
Recv:  0 -0.07500 +0.02500 -0.07500
Recv:  1 -0.02500 +0.02500 -0.12500
Recv:  2 -0.02500 -0.05000 -0.25000
Recv: 
Recv: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
Recv: ok
```

token47, 2018-04-22:
Yes! It works now. The G29 T is sent correctly and after the bed leveling the graph is plotted.

A sugestion: the Data Collector Flag comes empty (you are supposed to fill in your message, I understand) but because of the help text that is in the background inside of the text box, it seems that it's already filled. It took me a while to get that I could type over that.

jneilliii, 2018-04-22:
That's understandable. I'm using the placeholder attribute of the input field to try and show examples of what's supposed to go in there. Adding it as text below may make it clearer. The input field was originally a select list but there ended up being so many variations of flags to use it made more sense to make it open. Thinking of adding an initial wizard to step through the settings with instructions.

closed 2018-04-22 by jneilliii

jslay88, 2018-04-23:
Would probably be best to compile a list of the most common, and autocheck for those. Then, if need be, a user can specify their exact flag. Would provide a better 'out-of-the-box' experience with really no additional overhead. Could even store the known flag once auto detected the first time, if you really wanted to limit the steps beyond discovery.