jneilliii/OctoPrint-BedLevelVisualizer #10: Just spins...never shows a graph

CaffGeek, 2018-04-17:
I see the following commands get run in the terminal, with this output, but the plugin screen just spins, never updating or displaying anything

```
Send: G28
Send: G29 T1
Send: M503
Recv: ok
[...]
Recv: -2.130 -2.130 -2.130 -2.140 -2.190 -2.190 -2.190
Recv: -0.840 -0.840 -0.560 -0.600 -0.920 -1.460 -1.460
Recv: -0.130 0.380 0.670 0.610 0.220 -0.400 -1.200
Recv: 0.850 1.310 1.560 1.510 1.080 0.430 -0.470
Recv: 1.540 1.900 2.070 1.930 1.510 0.770 -0.080
Recv: 1.990 1.990 2.070 1.920 1.420 0.650 0.650
Recv: 1.550 1.550 1.550 1.320 0.910 0.910 0.910
Recv: ok
[...]
Recv: echo:Steps per unit:
Recv: echo:  M92 X100.00 Y100.00 Z100.00 E160.00
Recv: echo:Maximum feedrates (mm/s):
Recv: echo:  M203 X200.00 Y200.00 Z200.00 E200.00
Recv: echo:Maximum Acceleration (mm/s2):
Recv: echo:  M201 X4000 Y4000 Z4000 E4000
Recv: echo:Acceleration: S=acceleration, T=retract acceleration
Recv: echo:  M204 S2500.00 T3000.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X20.00 Z20.00 E20.00
Recv: echo:Home offset (mm):
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:PID settings:
Recv: echo:   M301 P24.77 I1.84 D83.61
Recv: ok
```

thomasnordquist, 2018-04-17:
Have similar problem, at least the title matches.
Using OctoPrint 1.3.8 and Marlin 1.1.8
I am using the AUTO_BED_LEVELING_LINEAR function with a 3x3 matrix.

[image]
```
Recv: echo:busy: processing
Recv: 
Recv: Bed Height Topography:
Recv:    +--- BACK --+
Recv:    |           |
Recv:  L |    (+)    | R
Recv:  E |           | I
Recv:  F | (-) N (+) | G
Recv:  T |           | H
Recv:    |    (-)    | T
Recv:    |           |
Recv:    O-- FRONT --+
Recv:  (0,0)
Recv:  -0.03217 +0.13583 +0.10433
Recv:  -0.12017 -0.02367 -0.04217
Recv:  -0.01767 +0.00783 -0.01217
Recv: 
Recv: X:144.01 Y:200.01 Z:11.21 E:0.00 Count X:14400 Y:20000 Z:4520
Recv: ok
```

CaffGeek, 2018-04-17:
I did notice our output is slightly different, it doesn't contain commas like the README shows

ButchMonkey, 2018-04-17:
I am also encountering this issue :

```
Send: G29 T1
Recv: Bed x: 0.00 y: 0.00 z: -2.86
Recv: Bed x: 90.00 y: 0.00 z: -3.32
Recv: Bed x: 180.00 y: 0.00 z: -3.65
Recv: Bed x: 180.00 y: 90.00 z: -2.78
Recv: Bed x: 90.00 y: 90.00 z: -2.37
Recv: Bed x: 0.00 y: 90.00 z: -2.16
Recv: Bed x: 0.00 y: 180.00 z: -1.38
Recv: Bed x: 90.00 y: 180.00 z: -1.98
Recv: Bed x: 180.00 y: 180.00 z: -1.87
Recv: Eqn coefficients: a: -0.00 b: 0.01 d: -2.93
Recv: planeNormal x: 0.00 y: -0.01 z: 1.00
```

jneilliii, 2018-04-17:
Yes, that's correct. Your firmware must respond with something that identifies the next return lines are leveling data. Some of these I can add, but other examples I've seen doesn't look like it reports anything prior to the data.

@ButchMonkey: what is your firmware and abl method? I've not seen that type of report.

CptanPanic, 2018-04-17:
I have Marlin firmware and get output like @thomasnordquist  There are lots of options for G29 so maybe it is a matter of choosing the correct ones?  http://marlinfw.org/docs/gcode/G029-abl.html

CaffGeek, 2018-04-17:
Yeah, Marlin here too, not sure how to check if I have UBL, or how to enable it

jneilliii, 2018-04-17:
Ok guys, just released the next iteration.  Made the "Data Colector Flag" setting an open text box.  Once you upgrade, you will need to manually modify the text to meet your needs.

@thomasnordquist : you should use `(0,0)` it seems.
@CaffGeek: doesn't look like you have any text just before the mesh data, so not sure this will work for you.  It's hard to tell though because your terminal is filtered and the last command sent looks like M503.  You may be experiencing the issue others have reported with a tempsgraph plugin conflict, see #9.
@CptanPanic: hopefully the open text field will work in your case.
@ButchMonkey: still no love for you, I think maybe with different T# options on the G29 command you might be able to get one that has a format this plugin likes.

closed 2018-04-17 by jneilliii

ButchMonkey, 2018-04-18:
@jneilliii My Marlin firmware is something like 2 years out of date - I'm using AUTO_BED_LEVELING_GRID.
Will probably just work on porting my config over to a newer version

CaffGeek, 2018-04-18:
@ButchMonkey Mine is pretty old to. Probably just going to update the FW to latest, and see what happens

CaffGeek, 2018-04-19:
@jneilliii I've updated my marlin firmware, and the bed visualizer, I now get output like this, and selected 
Data Collector Flag: `Bed Topography Report:` but still no luck

```Send: G29 T1
Recv: 
Recv: Bed Topography Report for CSV:
Recv: 
Recv: 0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000
Recv: 0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000
Recv: 0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000
Recv: 0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000
Recv: 0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000
Recv: 0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000
Recv: 0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000
Recv: 0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000
Recv: 0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000
Recv: 0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000
Recv: ok```

jneilliii, 2018-04-19:
The latest version of this plugin opened up the Data Collector Flag setting to be an open ended text field now. Make sure that settings matches whatever text you see just prior to the mesh data.  In the above case that would be 

> Bed Topography Report for CSV:

Once you have that in the settings it should work.

jneilliii, 2018-07-29:
@ButchMonkey: I have a test version ready for you to test, if you're willing and still on the older firmware version.  Install using the following url in Plugin Manager and let me know how it works out.  The install will take a little time as it will need to install an additional python library, just wait out the install, it will finish.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.3.zip