jneilliii/OctoPrint-BedLevelVisualizer #11: Let users customise "Data Collector Flag"

ysoldak, 2018-04-17:
Many printers, many flags.
Just let people define totally custom flag message instead of pre-defined drop-down.

I can workaround this limitation by manually injecting one more option via dev tools in my browser, but that's a hassle. And supporting all the flags that exist out there in the plugin is not an option, i guess (pardon the pun).

Example of the flag, for Monoprice Mini Delta:
`G29 Auto Bed Leveling`

Plugin version installed: 0.0.3

Thank you for the plugin.

jneilliii, 2018-04-17:
I agree, that's what I was about to do.

closed 2018-04-17 by jneilliii

jneilliii, 2018-04-17:
New version released.

ysoldak, 2018-04-18:
It works, thank you. Installed 0.0.4 and verified.

joelhuebner, 2019-02-20:
0.0.4 is for a Mini Delta, how about the Mini IIIp?

jneilliii, 2019-02-21:
I don't know how that device reports. If it's a marlin variant it should work.