jneilliii/OctoPrint-BedLevelVisualizer #8: Don't see result on CR-10S Marlin 1.1.8 BLtouch

ShagoY, 2018-04-17:
Hi :)
After a lot of try (G29/G29 T1/etc..) , i don't see result on bedvisualizer tab.

Here screenshot : 
[image]
[image]

jneilliii, 2018-04-17:
Looks like the data is getting extra information. Cam you send the terminal tab results of the command G29. Also did you try swapping the G29 command with G29 T?

Desterly, 2018-04-17:
@ShagoY do you have any other plugins that utilize plotly installed?  I just ran across a conflict with the Tempsgraph plugin that results in the same issues you are seeing.   If you open your browser debugger, see if you are getting any errors on the console when selecting the bedlevelvisualizer tab.  

#9

jneilliii, 2018-04-17:
Outside of the tempsgraph conflict, I just released a new version that will allow you to enter your own text. @ShagoY, you would use the text `Subdivided with CATMULL ROM Leveling Grid:` in that field, but until the conflict is resolved, I don't think we're going to get anywhere for you guys.

closed 2018-04-17 by jneilliii