jneilliii/OctoPrint-BedLevelVisualizer #20: Only for Cartesian printers??

mylife4aiurr, 2018-04-21:
just tried your pluggin. I think it a cool idea to show the data visually. 
Got everything working but I dont see my data represented. 

I have a delta printer. The visualization graphic shown on the "bed visualizer" model 
tab for me shows an empty x/y/z grid. 
That seems like only for Cartesian printers. idk if delta was not considered or not supported?

If u do plan to support Delta's here my ubl info:
```
Send: M420 V
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv: (0,9)                                                                   (9,9)
Recv: (-98,98)                                                                (98,98)
Recv:      .          .          .          .       0.100    0.100       .          .          .          .
Recv: 
Recv:      .          .       0.155    0.205    0.245    0.235    0.225    0.225       .          .
Recv: 
Recv:      .       0.140    0.190    0.195    0.265    0.300    0.295    0.275    0.325       .
Recv: 
Recv:      .       0.140    0.150    0.200    0.285    0.310    0.305    0.300    0.320       .
Recv: 
Recv:   0.050    0.110    0.165    0.190    0.280    0.295    0.325    0.345    0.345    0.150
Recv: 
Recv:      .       0.074    0.250    0.275  [ 0.325]   0.310    0.300    0.385    0.395    0.150
Recv: 
Recv:      .       0.050    0.154    0.231    0.229    0.206    0.330    0.400    0.450       .
Recv: 
Recv:      .       0.200    0.128    0.159    0.186    0.305    0.370    0.440    0.525       .
Recv: 
Recv:      .          .       0.201    0.186    0.221    0.255    0.311    0.538       .          .
Recv: 
Recv:      .          .          .          .       0.200    0.245       .          .          .          .
Recv: (-98,-98)                                                                    (98,-98)
Recv: (0,0)                                                                     (9,0)
Recv: ubl.mesh_is_valid = 0
Recv: ubl.storage_slot = 1
Recv: echo:Bed Leveling On
Recv: ok P15 B3
```

jneilliii, 2018-04-21:
Did you set the Data Collector Flag in settings?  In your case it would need to be set to `Bed Topography Report:`

jneilliii, 2018-04-21:
I'll have to look to see if there is a way to plot the surface with those missing points, relative to the cartesian grid.

mylife4aiurr, 2018-04-21:
I was mistaken B4. The map isn't blank. It is however only pulling the middle line of data points from my topography report:
 0.050    0.110    0.165    0.190    0.280    0.295    0.325    0.345    0.345    0.150
All other data points above and below middle line are not rendered and or disreguarded and not graphed.

In your example pic. Under saved mesh points, I see many lines of point data. For my Delta printer in saved mesh I only see one line of data points. 

So yea have to get interpretor to understand circle of data points not only rectangle/square

mylife4aiurr, 2018-04-21:
If I use:
G29 T1 instead of M420 V
output looks like this:
```
Recv: Bed Topography Report for CSV:
Recv: 
Recv: NAN,NAN,NAN,NAN,0.100,0.100,NAN,NAN,NAN,NAN
Recv: NAN,NAN,0.155,0.205,0.245,0.235,0.225,0.225,NAN,NAN
Recv: NAN,0.140,0.190,0.195,0.265,0.300,0.295,0.275,0.325,NAN
Recv: NAN,0.140,0.150,0.200,0.285,0.310,0.305,0.300,0.320,NAN
Recv: 0.050,0.110,0.165,0.190,0.280,0.295,0.325,0.345,0.345,0.150
Recv: NAN,0.074,0.250,0.275,0.325,0.310,0.300,0.385,0.395,0.150
Recv: NAN,0.050,0.154,0.231,0.229,0.206,0.330,0.400,0.450,NAN
Recv: NAN,0.200,0.128,0.159,0.186,0.305,0.370,0.440,0.525,NAN
Recv: NAN,NAN,0.201,0.186,0.221,0.255,0.311,0.538,NAN,NAN
Recv: NAN,NAN,NAN,NAN,0.200,0.245,NAN,NAN,NAN,NAN

```
When I use G29 T1 i change data collector flag to 

> Bed Topography Report for CSV:

but still wont get graphed by your pluggin.

mylife4aiurr, 2018-04-21:
Follow up: 
if I manually enter my data points at https://plot.ly/create/, the site your using its able to graph the points.

jneilliii, 2018-04-21:
I think I could get this report to work, but have to account for the NAN text.  When you are entering it on the plotly website, how are you entering the points in the grid?  I tried and got some skewed results.

[image]

mylife4aiurr, 2018-04-21:
Since I was manually entering the data, I only entered actual data. example 1st line with M420: 
  .          .          .          .       0.100    0.100       .          .          .          .
@ website I left empty cells with a "." -> blank in spreadsheet
if I had used G29 T1 with csv output "NAN", -> blank, in spreadsheet
NAN,NAN,NAN,NAN,0.100,0.100,NAN,NAN,NAN,NAN

Then I just kept entering the data
[image]

jneilliii, 2018-04-22:
Try installing the plugin from the link below and let me know if that works for you or not.  After install make sure the web interface reloads, otherwise press ctrl+f5.  Theoretically it should work.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.0.8.zip

mylife4aiurr, 2018-04-22:
[image]
[image]

Notice stored mesh data..... its only pulling middle line of data points

jneilliii, 2018-04-22:
I've made some adjustments to the same branch linked before.  Give it another try and let me know.

mylife4aiurr, 2018-04-22:
Working success using- 
data collector flag>
Bed Topography Report:

GCODE Commands>
G28
M420 V or G29 T

[image]
Graph was rendered like website :) 

However the "saved mesh data" field still only shows 1 line of data NOT whole data set like picture from your example.
[image]

-------------------------------------------------------
Not working
data collector flag>
Bed Topography Report for CSV:

GCODE Commands>
G28
G29 T1
```
Send: G29 T1
Recv: 
Recv: Bed Topography Report for CSV:
Recv: 
Recv: NAN,NAN,NAN,NAN,0.100,0.100,NAN,NAN,NAN,NAN
Recv: NAN,NAN,0.155,0.205,0.245,0.235,0.225,0.225,NAN,NAN
Recv: NAN,0.140,0.190,0.195,0.265,0.300,0.295,0.275,0.325,NAN
Recv: NAN,0.140,0.150,0.200,0.285,0.310,0.305,0.300,0.320,NAN
Recv: 0.050,0.110,0.165,0.190,0.280,0.295,0.325,0.345,0.345,0.150
Recv: NAN,0.074,0.250,0.275,0.325,0.310,0.300,0.385,0.395,0.150
Recv: NAN,0.050,0.154,0.231,0.229,0.206,0.330,0.400,0.450,NAN
Recv: NAN,0.200,0.128,0.159,0.186,0.305,0.370,0.440,0.525,NAN
Recv: NAN,NAN,0.201,0.186,0.221,0.255,0.311,0.538,NAN,NAN
Recv: NAN,NAN,NAN,NAN,0.200,0.245,NAN,NAN,NAN,NAN
```
[image]

-------------------------------------------------------
The rendering of the graph has worked using gcode that doesnt outputs a csv format (M420 V or G29 T). 

If i use G29 T1, which outputs csv format - graph not rendered.

On settings page whole data set not shown to user, just the single line. Whole data set not shown if csv format or human readable.

mylife4aiurr, 2018-04-22:
From config.yaml
```
    camera: true
  bedlevelvisualizer:
    command: 'G28

      G29 T

      '
    mesh_timestamp: 4/21/2018, 11:44:16 PM
    report_flag: 'Bed Topography Report:'
    save_mesh: false
    stored_mesh:
    - - '0.050'
      - '0.110'
      - '0.165'
      - '0.190'
      - '0.280'
      - '0.295'
      - '0.325'
      - '0.345'
      - '0.345'
      - '0.150'
```

jneilliii, 2018-04-22:
Thanks for all the testing.  I didn't take into account the CSV mode, will have to merge that back in.  As for the stored mesh, the reason it's not showing the correct data is because you don't have saving enabled, so it's still showing the last data that it got when that option was enabled.  Re-enable saving mesh in settings and press the update button on the visualizer tab and the data will get updated in config.yaml.

jneilliii, 2018-04-22:
Ok, I think I've got all use cases working now baesd on all the different reports that I was seeing posted in the issues.  Should now work with the CSV export as well. In theory, should work with all variations.  If you don't mind giving it some testing for me that would be greatly appreciated prior to me releasing an updated version.

jneilliii, 2018-04-22:
Hold off on that, csv still isn't working correct.

mylife4aiurr, 2018-04-22:
I dont mind testing..... but leave open issue unless fixed?

Dont u wanna separate issues?
One if data is being displayed, if human readable or csv for i3 style and delta

One for changing how data is displayed to user if delta (hoping for a circle bed representation) #28

mylife4aiurr, 2018-04-22:
General for delta, not just mine.
Pull data from Topography Report
(0,9)        (9,9)
(-98,98) **_min_**    (98,98)  **_max_**
that will have actual ranges
-(1/2min)  x(0)  +(1/2max)
-(1/2min)  y(0)  +(1/2max)

what do you think?

mylife4aiurr, 2018-04-22:
ok i'll try it now

mylife4aiurr, 2018-04-22:
Confirmed regular and csv output confirmed displaying to user for me as well

jneilliii, 2018-04-23:
Resolved with release of version 0.0.8

closed 2018-04-23 by jneilliii

jneilliii, 2018-04-26:
Would you guys be willing to test the 0.0.9 branch from the url below on your deltas?  I'm looking into simplifying the data collector and I think it might break it for you, but wanted to double check.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.0.9.zip

mylife4aiurr, 2018-04-27:
I can do some testing tomorrow night and this weekend. Was planning on updating to lastest bugfix 1.1.x anyway.

mylife4aiurr, 2018-04-28:
Finished updating to newer marlin bugfix 1.1.x commit (b4ddee8bebead400142fe45edc3bc41bea4ecb16), only for a newer commit (3ca3268241b9ce6040ccf413556aa2cbb0fe3bac) to be released a few hours later. 

But the point I'm making is the output of gcode commands M420 / G29 T has changed, as pointed in #40. The change breaks your pluggin for me.

However the csv format still works with your pluggin. 
So to make it work I used:
G28
G29 T1
Bed Topography Report for CSV:

The flipping x or y functionality also worked

Tested on version 0.0.9.

jneilliii, 2018-04-28:
I thought that the change I made on the 0.0.9 branch would work with the format posted on #40, but I guess not.  I'm abandoning that change any way and will adjust the previous 0.0.8 code to work with the newer format...looks like I'll be flashing, etc. I have made improvements already to the graph for #28.