jneilliii/OctoPrint-BedLevelVisualizer #22: Add support for G32 in Smoothieware?

sixdeuces, 2018-04-21:
Looks like an excellent little addition to Octoprint! I'd really like to make use of this on my delta, but Smoothie uses G31 instead of G29. Any chance we can get that support added as an option? Thank you!

jneilliii, 2018-04-21:
You can set the gcode to whatever you want in settings. You also need to set appropriate data collector flag.

jneilliii, 2018-04-25:
From what I can tell from the message that appeared in my email, but not here is if you were to use `Delta Grid Probe...` in the data collector flag setting it may work assuming there is an ok response after the probe complete message.  That's what tells the collector that the data is completely listed. If Smoothieware doesn't respond with an ok, it looks like you could add `M400` to force it to respond ok, maybe.

Duckle29, 2018-04-26:
Here's a complete probe on my delta running smoothieware:
```
Send: G31
Recv: Delta Grid Probe...
Recv: Probe start ht is 19.699982 mm, probe radius is 200.000000 mm, grid size is 11x11
Recv: probe at 0,0 is 0.000000 mm
Recv: DEBUG: X0.0000, Y-200.0000, Z-0.5800
Recv: DEBUG: X-120.0000, Y-160.0000, Z-0.3300
Recv: DEBUG: X-80.0000, Y-160.0000, Z-0.4600
Recv: DEBUG: X-40.0000, Y-160.0000, Z-0.4200
Recv: DEBUG: X0.0000, Y-160.0000, Z-0.4000
Recv: DEBUG: X40.0000, Y-160.0000, Z-0.4500
Recv: DEBUG: X80.0000, Y-160.0000, Z-0.3400
Recv: DEBUG: X120.0000, Y-160.0000, Z-0.2800
Recv: DEBUG: X160.0000, Y-120.0000, Z-0.5000
Recv: DEBUG: X120.0000, Y-120.0000, Z-0.3700
Recv: DEBUG: X80.0000, Y-120.0000, Z-0.2800
Recv: DEBUG: X40.0000, Y-120.0000, Z-0.3200
Recv: DEBUG: X0.0000, Y-120.0000, Z-0.2600
Recv: DEBUG: X-40.0000, Y-120.0000, Z-0.2700
Recv: DEBUG: X-80.0000, Y-120.0000, Z-0.3100
Recv: DEBUG: X-120.0000, Y-120.0000, Z-0.3700
Recv: DEBUG: X-160.0000, Y-120.0000, Z-0.3500
Recv: DEBUG: X-160.0000, Y-80.0000, Z-0.3300
Recv: DEBUG: X-120.0000, Y-80.0000, Z-0.2700
Recv: DEBUG: X-80.0000, Y-80.0000, Z-0.2100
Recv: DEBUG: X-40.0000, Y-80.0000, Z-0.1300
Recv: DEBUG: X0.0000, Y-80.0000, Z-0.1200
Recv: DEBUG: X40.0000, Y-80.0000, Z-0.1600
Recv: DEBUG: X80.0000, Y-80.0000, Z-0.1300
Recv: DEBUG: X120.0000, Y-80.0000, Z-0.2300
Recv: DEBUG: X160.0000, Y-80.0000, Z-0.4300
Recv: DEBUG: X160.0000, Y-40.0000, Z-0.3700
Recv: DEBUG: X120.0000, Y-40.0000, Z-0.2000
Recv: DEBUG: X80.0000, Y-40.0000, Z-0.1500
Recv: DEBUG: X40.0000, Y-40.0000, Z-0.0500
Recv: DEBUG: X0.0000, Y-40.0000, Z-0.0800
Recv: DEBUG: X-40.0000, Y-40.0000, Z-0.0400
Recv: DEBUG: X-80.0000, Y-40.0000, Z-0.0700
Recv: DEBUG: X-120.0000, Y-40.0000, Z-0.1200
Recv: DEBUG: X-160.0000, Y-40.0000, Z-0.1500
Recv: DEBUG: X-200.0000, Y0.0000, Z0.1000
Recv: DEBUG: X-160.0000, Y0.0000, Z-0.1000
Recv: DEBUG: X-120.0000, Y0.0000, Z-0.0600
Recv: DEBUG: X-80.0000, Y0.0000, Z-0.0400
Recv: DEBUG: X-40.0000, Y0.0000, Z-0.0200
Recv: DEBUG: X0.0000, Y0.0000, Z0.0700
Recv: DEBUG: X40.0000, Y0.0000, Z0.0500
Recv: DEBUG: X80.0000, Y0.0000, Z-0.0200
Recv: DEBUG: X120.0000, Y0.0000, Z-0.1700
Recv: DEBUG: X160.0000, Y0.0000, Z-0.2900
Recv: DEBUG: X200.0000, Y0.0000, Z-0.3400
Recv: DEBUG: X160.0000, Y40.0000, Z-0.4200
Recv: DEBUG: X120.0000, Y40.0000, Z-0.1900
Recv: DEBUG: X80.0000, Y40.0000, Z-0.0300
Recv: DEBUG: X40.0000, Y40.0000, Z-0.0200
Recv: DEBUG: X0.0000, Y40.0000, Z0.0600
Recv: DEBUG: X-40.0000, Y40.0000, Z0.1100
Recv: DEBUG: X-80.0000, Y40.0000, Z0.1200
Recv: DEBUG: X-120.0000, Y40.0000, Z0.0600
Recv: DEBUG: X-160.0000, Y40.0000, Z-0.0100
Recv: DEBUG: X-160.0000, Y80.0000, Z0.0500
Recv: DEBUG: X-120.0000, Y80.0000, Z0.0900
Recv: DEBUG: X-80.0000, Y80.0000, Z0.0700
Recv: DEBUG: X-40.0000, Y80.0000, Z0.1000
Recv: DEBUG: X0.0000, Y80.0000, Z0.1100
Recv: DEBUG: X40.0000, Y80.0000, Z0.0500
Recv: DEBUG: X80.0000, Y80.0000, Z-0.0500
Recv: DEBUG: X120.0000, Y80.0000, Z-0.2000
Recv: DEBUG: X160.0000, Y80.0000, Z-0.3800
Recv: DEBUG: X160.0000, Y120.0000, Z-0.4600
Recv: DEBUG: X120.0000, Y120.0000, Z-0.2200
Recv: DEBUG: X80.0000, Y120.0000, Z-0.1000
Recv: DEBUG: X40.0000, Y120.0000, Z0.0700
Recv: DEBUG: X0.0000, Y120.0000, Z0.0800
Recv: DEBUG: X-40.0000, Y120.0000, Z0.1500
Recv: DEBUG: X-80.0000, Y120.0000, Z0.1700
Recv: DEBUG: X-120.0000, Y120.0000, Z0.2100
Recv: DEBUG: X-160.0000, Y120.0000, Z0.2400
Recv: DEBUG: X-120.0000, Y160.0000, Z0.4100
Recv: DEBUG: X-80.0000, Y160.0000, Z0.3200
Recv: DEBUG: X-40.0000, Y160.0000, Z0.1300
Recv: DEBUG: X0.0000, Y160.0000, Z0.0800
Recv: DEBUG: X40.0000, Y160.0000, Z0.0200
Recv: DEBUG: X80.0000, Y160.0000, Z-0.0500
Recv: DEBUG: X120.0000, Y160.0000, Z-0.2200
Recv: DEBUG: X0.0000, Y200.0000, Z0.1500
Recv: -0.3200 -0.3900 -0.5900 -0.5800 -0.5699 -0.5800 -0.5800 -0.5799 -0.3600 -0.2800 -0.7700
Recv: -0.4300 -0.3700 -0.3300 -0.4600 -0.4200 -0.4000 -0.4500 -0.3400 -0.2800 -0.5700 -0.7700
Recv: -0.5400 -0.3500 -0.3700 -0.3100 -0.2700 -0.2600 -0.3200 -0.2800 -0.3700 -0.5000 -0.6600
Recv: -0.3900 -0.3300 -0.2700 -0.2100 -0.1300 -0.1200 -0.1600 -0.1300 -0.2300 -0.4300 -0.6300
Recv: -0.1800 -0.1500 -0.1200 -0.0700 -0.0400 -0.0800 -0.0500 -0.1500 -0.2000 -0.3700 -0.5400
Recv:  0.1000 -0.1000 -0.0600 -0.0400 -0.0200  0.0700  0.0500 -0.0200 -0.1700 -0.2900 -0.3400
Recv: -0.0800 -0.0100  0.0600  0.1200  0.1100  0.0600 -0.0200 -0.0300 -0.1900 -0.4200 -0.6500
Recv:  0.0100  0.0500  0.0900  0.0700  0.1000  0.1100  0.0500 -0.0500 -0.2000 -0.3800 -0.6700
Recv:  0.1000  0.2400  0.2100  0.1700  0.1500  0.0800  0.0700 -0.1000 -0.2200 -0.4600 -0.6899
Recv:  0.3900  0.4300  0.4100  0.3200  0.1300  0.0800  0.0200 -0.0500 -0.2200 -0.3900 -0.7099
Recv:  0.6799  0.6500  0.4900  0.1800  0.1100  0.1500 -0.0300 -0.0399 -0.1700 -0.3200 -0.5601
Recv: Probe completed - Enter M374 to save this grid
Recv: ok
```

```
Build version: edge-c819aa6, Build date: Apr 4 2018 16:52:08, MCU: LPC1769, System Clock: 120MHz 
5 axis
```
So it does respond with an ok, but it also spams a lot of other info. Would this work with smoothieware?

jneilliii, 2018-04-26:
Try using `Delta Grid Probe...` in the data collector setting.

Duckle29, 2018-04-26:
@jneilliii I did that, and it does seem to kinda work: https://i.imgur.com/mBWnOjl.png

However being a delta, where would my 0,0 be on that?

jneilliii, 2018-05-02:
New release 0.1.0 might fix this display issue, may make it worse for you.  Give it a try and let me know.

jneilliii, 2018-06-05:
Any updates on this ticket?  I'll close tomorrow if I don't hear back from you.

Thisismydigitalself, 2018-06-08:
I am sorry for not speaking the programmer's language. 
In layman's terms: what should an ignoramus like myself enter in "GCODE Commands" text field in order to get this beautiful height map of my Smoothie powered delta. I ran G31 before each print.

Thank you.

jneilliii, 2018-06-09:
In theory you should be able to enter the following, but no-one has yet to report back.  If you tell me what you get in response to a `G31` command in the terminal tab I could give a little better guidance.

Thisismydigitalself, 2018-06-10:
Thank you. 
The following is Smoothie's reply to G31 command:

Send: N124 G31*28
Recv: Delta Grid Probe...
Recv: Probe start ht is -4.030304 mm, probe radius is 140.000000 mm, grid size is 5x5
Recv: probe at 0,0 is 0.022552 mm
Recv: DEBUG: X0.0000, Y-140.0000, Z0.0753
Recv: DEBUG: X-70.0000, Y-70.0000, Z-0.0439
Recv: DEBUG: X0.0000, Y-70.0000, Z0.0188
Recv: DEBUG: X70.0000, Y-70.0000, Z0.1943
Recv: DEBUG: X140.0000, Y0.0000, Z0.0941
Recv: DEBUG: X70.0000, Y0.0000, Z0.2131
Recv: DEBUG: X0.0000, Y0.0000, Z-0.0063
Recv: DEBUG: X-70.0000, Y0.0000, Z-0.0627
Recv: DEBUG: X-140.0000, Y0.0000, Z-0.3886
Recv: DEBUG: X-70.0000, Y70.0000, Z-0.1316
Recv: DEBUG: X0.0000, Y70.0000, Z0.0878
Recv: DEBUG: X70.0000, Y70.0000, Z0.2257
Recv: DEBUG: X0.0000, Y140.0000, Z0.0439
Recv: -0.0815 -0.0251  0.0753  0.1755  0.3949
Recv: -0.1066 -0.0439  0.0188  0.1943  0.3698
Recv: -0.3886 -0.0627 -0.0063  0.2131  0.0941
Recv: -0.3510 -0.1316  0.0878  0.2257  0.3636
Recv: -0.3134 -0.2006  0.0439  0.2382  0.4576
Recv: Probe completed
Recv: ok
[...]

jneilliii, 2018-06-10:
The following should in theory work with the latest version of the plugin. As mentioned before in this issue thread; the only downside is the plugin will utilize the last part of that report without the DEBUG lines and generate a rectangular grid, not a round delta style surface like you would get with Marlin.

```
@BELEVELVISUALIZER
G31
```

Thisismydigitalself, 2018-06-11:
Thank you.
I got a plot. not sure what I can make of it.
If I understand it correctly, I have ~1mm slant from left to right on a 340mm builtplate which I should try and mechanically correct for as flat & level bed as possible. 

[image]

[image]

Thisismydigitalself, 2018-06-11:
I like this plugin.
I believe i managed to use it the right way and physically corrected my bed level.

[image]
[image]

jneilliii, 2018-06-11:
Glad that worked.  Now that you have confirmed it is working for Smoothieware I'm closing this issue.

closed 2018-06-11 by jneilliii