jneilliii/OctoPrint-BedLevelVisualizer #23: Please wait, retrieving current mesh.

J-charles-C, 2018-04-21:
I,

I don't have the graph after the G29T.

Wath is the problem ?
[image]

best

jneilliii, 2018-04-21:
I'd have to see the standard results you get when issuing a G29 T to tell you.  More than likely, you just have the Data Collector Flag setting incorrect relative to your report output.

4ntani, 2018-04-21:
Hi, 

thank you so much for have make this plug-in!  
But i think there are some problems with the data transmission.

This is the output printed by the terminal:

Send: G28
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: clean up finished
Recv: babystep applied
Recv: Bed leveling correction finished
Recv: Upsample finished
Recv: Mesh bed leveling activated
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Go home finished
Recv: ok
Send: G29 T
Recv: ok

[image]

[image]

I have an Original Prusa Mk2s with the lastest firmware... What i should try do?

J-charles-C, 2018-04-21:
I,

Recv: echo:busy: processing
Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 -0.187 +0.015 +0.198
Recv:  1 -0.176 +0.000 +0.059
Recv:  2 -0.260 -0.274 -0.203
Recv: 
Recv: X:182.00 Y:199.00 Z:16.38 E:0.00 Count A:30594 B:-1394 Z:6429
Recv: ok

jneilliii, 2018-04-21:
@4ntani, replace the gcode script commands in the plugin's settings and enable prusa firmware option. Make sure to set the data collector flag to `Measured points:`.

jneilliii, 2018-04-21:
@J-charles-C, everything looks correct. Are there any errors in browser console or octoprint.log?

tonyccol, 2018-04-21:
@jneilliii @4ntani Replace the gcode script commands with what? i have the MK2s as well and am trying to get this plugin to work

J-charles-C, 2018-04-21:
When i install or uninstall the plugin, i have this message 
[image]

Log and terminal are OK.

In terminal : -bash: pip install --upgrade pip: command not found

best

J-charles-C, 2018-04-21:
Ok, for PIP update : 

/home/pi/oprint/bin/python -m pip install --upgrade pip

best

jneilliii, 2018-04-21:
@4ntani: how about usingthe consumers for bed leveling (G80) and bed leveling (G81)?

jneilliii, 2018-04-21:
@J-charles-C, pip is unrelated to the plugin issue you are having.  Looks like you tried uninstalling re-installing.  Do you see any errors in your ocotprint.log or browser's console when you try to update the mesh or switch to the tab?

J-charles-C, 2018-04-21:
[octoprint.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/1934790/octoprint.log)
No error.

best

jneilliii, 2018-04-21:
That's really strange then.  Potentially a localization issue.  Can you try going to the main page (not in this plugins tab) and press CTRL+F5. Once reloaded press F12 to open developer tools and then on the console tab in developer tools active switch to this plugin's tab.  Are there any errors shown in the console?  It will also spit out write out the mesh data there once it's detected.

4ntani, 2018-04-22:
@jneilliii @tonyccol with G81 everything works!!!!

[image]

[image]

Thank you so much for the help!!!! Keep it up!

J-charles-C, 2018-04-26:
I jneilliii,

With the last version, i have this message :
TypeError: undefined is not an object (evaluating 's[0].length')

Erase cache and other test don't resolve my problem

best

ytymec, 2018-04-26:
J-charles-C , if you are using Marlin or a similare bord, install the latest version solved the problem on my side.

J-charles-C, 2018-04-26:
I, i have the lastest marlin 1.1.8

best

jneilliii, 2018-04-26:
Can you upload the plugin settings stored in config.yaml?

J-charles-C, 2018-04-28:
I,

`  bedlevelvisualizer:
    command: 'M155 S30

      G28

      G29

      M155 S3'
    mesh_timestamp: 20/04/2018 à 20:20:51
    report_flag: 'Bilinear Leveling Grid: '`

best

J-charles-C, 2018-04-29:
I,

I have cut the space after the ":". No more success

best

jneilliii, 2018-04-29:
@J-charles-C, I have made some significant improvements in the data collector process in my upcoming 0.1.0 release that eliminates the need for the data collector flag and may have better results for you as well,  Can you install the plugin from the URL below and report back if there is any change for you? If not please copy/paste the terminal output after pressing the update button.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.0.zip

J-charles-C, 2018-04-29:
I,

Install 0.1.0

Installation du plugin depuis https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.0.zip...
/home/pi/oprint/bin/python -m pip install https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.0.zip
Collecting https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.0.zip
Downloading https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.0.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from Bed-Visualizer==0.1.0) (1.3.8)
Requirement already satisfied: flask<0.11,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (0.10.1)
Requirement already satisfied: Jinja2<2.9,>=2.8 in /home/pi/oprint/lib/python2.7/site-packages/Jinja2-2.8.1-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (2.8.1)
Requirement already satisfied: werkzeug<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/Werkzeug-0.8.3-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (0.8.3)
Requirement already satisfied: tornado==4.0.2 in /home/pi/oprint/lib/python2.7/site-packages/tornado-4.0.2-py2.7-linux-armv7l.egg (from OctoPrint->Bed-Visualizer==0.1.0) (4.0.2)
Requirement already satisfied: sockjs-tornado<1.1,>=1.0.3 in /home/pi/oprint/lib/python2.7/site-packages/sockjs_tornado-1.0.3-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (1.0.3)
Requirement already satisfied: PyYAML<3.11,>=3.10 in /home/pi/oprint/lib/python2.7/site-packages/PyYAML-3.10-py2.7-linux-armv7l.egg (from OctoPrint->Bed-Visualizer==0.1.0) (3.10)
Requirement already satisfied: Flask-Login<0.3,>=0.2.2 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Login-0.2.11-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (0.2.11)
Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Principal-0.3.5-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (0.3.5)
Requirement already satisfied: Flask-Babel<0.10,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Babel-0.9-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (0.9)
Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (0.10)
Requirement already satisfied: markdown<2.7,>=2.6.4 in /home/pi/oprint/lib/python2.7/site-packages/Markdown-2.6.8-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (2.6.8)
Requirement already satisfied: pyserial<3.5,>=2.7 in /home/pi/oprint/lib/python2.7/site-packages/pyserial-2.7-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (2.7)
Requirement already satisfied: netaddr<0.8,>=0.7.17 in /home/pi/oprint/lib/python2.7/site-packages/netaddr-0.7.19-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (0.7.19)
Requirement already satisfied: watchdog<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/watchdog-0.8.3-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (0.8.3)
Requirement already satisfied: sarge<0.2,>=0.1.4 in /home/pi/oprint/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (0.1.4)
Requirement already satisfied: netifaces<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/netifaces-0.10.6-py2.7-linux-armv7l.egg (from OctoPrint->Bed-Visualizer==0.1.0) (0.10.6)
Requirement already satisfied: pylru<1.1,>=1.0.9 in /home/pi/oprint/lib/python2.7/site-packages/pylru-1.0.9-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (1.0.9)
Requirement already satisfied: rsa<3.3,>=3.2 in /home/pi/oprint/lib/python2.7/site-packages/rsa-3.2.3-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (3.2.3)
Requirement already satisfied: pkginfo<1.3,>=1.2.1 in /home/pi/oprint/lib/python2.7/site-packages/pkginfo-1.2.1-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (1.2.1)
Requirement already satisfied: requests<3,>=2.18.4 in /home/pi/oprint/lib/python2.7/site-packages/requests-2.18.4-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (2.18.4)
Requirement already satisfied: semantic_version<2.5,>=2.4.2 in /home/pi/oprint/lib/python2.7/site-packages/semantic_version-2.4.2-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (2.4.2)
Requirement already satisfied: psutil==5.4.3 in /home/pi/oprint/lib/python2.7/site-packages/psutil-5.4.3-py2.7-linux-armv7l.egg (from OctoPrint->Bed-Visualizer==0.1.0) (5.4.3)
Requirement already satisfied: Click<6.3,>=6.2 in /home/pi/oprint/lib/python2.7/site-packages/click-6.2-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (6.2)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (1.6.5)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (5.2.1)
Requirement already satisfied: chainmap<1.1,>=1.0.2 in /home/pi/oprint/lib/python2.7/site-packages/chainmap-1.0.2-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (1.0.2)
Requirement already satisfied: future<0.16,>=0.15 in /home/pi/oprint/lib/python2.7/site-packages/future-0.15.2-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.0) (0.15.2)
Requirement already satisfied: scandir<1.4,>=1.3 in /home/pi/oprint/lib/python2.7/site-packages/scandir-1.3-py2.7-linux-armv7l.egg (from OctoPrint->Bed-Visualizer==0.1.0) (1.3)
[... cut ...]
Requirement already satisfied: pytz>=0a in /home/pi/oprint/lib/python2.7/site-packages/pytz-2017.2-py2.7.egg (from Babel>=1.0->Flask-Babel<0.10,>=0.9->OctoPrint->Bed-Visualizer==0.1.0) (2017.2)
Installing collected packages: Bed-Visualizer
Found existing installation: Bed-Visualizer 0.0.8
Uninstalling Bed-Visualizer-0.0.8:
Successfully uninstalled Bed-Visualizer-0.0.8
Running setup.py install for Bed-Visualizer: started
Running setup.py install for Bed-Visualizer: finished with status 'done'
Successfully installed Bed-Visualizer-0.1.0
Terminé !

J-charles-C, 2018-04-29:
Many Thanks for your help and this plugin !!!!!!

[image]

best

Jc

jneilliii, 2018-04-29:
That's great, the update succeeded, does the plugin work for you now?

jneilliii, 2018-04-29:
Awesome.  I'm waiting on some additional testing results prior to full release, but looks like this fixed it for you.

closed 2018-04-29 by jneilliii

J-charles-C, 2018-04-30:
I,

Mesh graphic don't update with the G29 data. I view only the first created. Control-F5 don't change that.

best

jneilliii, 2018-04-30:
So when you press the update button it doesn't retrieve the mesh and update, gets stuck on previously stored mesh?

J-charles-C, 2018-04-30:
Yes it's exactly my problem. 

And "Stored Mesh Data:" in parameters don't update too.

best

jneilliii, 2018-04-30:
Try this...if you have multiple commands in your GCODE setting, place the text `@BEDLEVELVISUALIZER` just above the `G29` command...like this.

[image]

What is happening is the collector stops when it receives an ok command, and if that flag isn't in the gcode you enter the ok is coming earlier than expected.

J-charles-C, 2018-05-01:
I,

Its Ok. I have  new mesh data and new graphic.

kwyatt242, 2018-07-25:
Great idea for a plugin!

Unfortunately, I seem to be stuck with perhaps a similar issue? The bed visualizer issues the commands but never receives any data. 

Info on my system:
Tronxy p802ma running Marlin 1.0
OctoPrint 1.3.9
Browser is Chrome 

This screen stays present after bed leveling completes...
[image]

Here are the commands I use...
[image]

This is what terminal is reporting...
[image]

Console log from Chrome...
[image]

Any help is much appreciated - thanks!

jneilliii, 2018-07-25:
Or looks like your printer is either using an old marlin version or smoothieware. The plugin unfortunately does not support the report that is coming back at this time.

kwyatt242, 2018-07-26:
Thank you - that makes sense and all the more confirms my decision to replace the controller board with something more modern.

jneilliii, 2018-07-27:
Some other users are getting that same report, #74.  I'm going to try and work this out for you guys this weekend.