jneilliii/OctoPrint-BedLevelVisualizer #25: Cannot install plugin

maddavo, 2018-04-21:
I get this when trying to install the plugin using the plugin manager:

> 

Installing plugin "Bed Level Visualizer" from https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip.../home/pi/oprint/bin/python -m pip install https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zipException:Traceback (most recent call last):File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in mainstatus = self.run(options, args)File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/commands/install.py", line 272, in runwith self._build_session(options) as session:File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/basecommand.py", line 72, in _build_sessioninsecure_hosts=options.trusted_hosts,File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 329, in __init__self.headers["User-Agent"] = user_agent()File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 93, in user_agentfrom pip._vendor import distroFile "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/distro.py", line 1050, in <module>_distro = LinuxDistribution()File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/distro.py", line 594, in __init__if include_lsb else {}File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/distro.py", line 933, in _get_lsb_release_inforaise subprocess.CalledProcessError(code, cmd, stdout)CalledProcessError: Command 'lsb_release -a' returned non-zero exit status 1Traceback (most recent call last):File "/usr/lib/python2.7/runpy.py", line 162, in _run_module_as_main"__main__", fname, loader, pkg_name)File "/usr/lib/python2.7/runpy.py", line 72, in _run_codeexec code in run_globalsFile "/home/pi/oprint/lib/python2.7/site-packages/pip/__main__.py", line 19, in <module>sys.exit(pip.main())File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/__init__.py", line 233, in mainreturn command.main(cmd_args)File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/basecommand.py", line 251, in maintimeout=min(5, options.timeout)) as session:File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/basecommand.py", line 72, in _build_sessioninsecure_hosts=options.trusted_hosts,File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 329, in __init__self.headers["User-Agent"] = user_agent()File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 93, in user_agentfrom pip._vendor import distroFile "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/distro.py", line 1050, in <module>_distro = LinuxDistribution()File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/distro.py", line 594, in __init__if include_lsb else {}File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/distro.py", line 933, in _get_lsb_release_inforaise subprocess.CalledProcessError(code, cmd, stdout)subprocess.CalledProcessError: Command 'lsb_release -a' returned non-zero exit status 1Error!Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

jneilliii, 2018-04-21:
What's in your plugin_pluginmanager_console.log file? I've not seen this error before from anyone else.

maddavo, 2018-04-21:
Pretty much as I have pasted except with timestamps...

2018-04-21 17:10:59,952   /home/pi/oprint/bin/python -m pip install https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip
2018-04-21 17:11:05,241 ! Exception:
2018-04-21 17:11:05,242 ! Traceback (most recent call last):
2018-04-21 17:11:05,243 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
2018-04-21 17:11:05,244 ! status = self.run(options, args)
2018-04-21 17:11:05,245 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/commands/install.py", line 272, in run
2018-04-21 17:11:05,246 ! with self._build_session(options) as session:
2018-04-21 17:11:05,246 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/basecommand.py", line 72, in _build_session
2018-04-21 17:11:05,247 ! insecure_hosts=options.trusted_hosts,
2018-04-21 17:11:05,248 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 329, in __init__
2018-04-21 17:11:05,248 ! self.headers["User-Agent"] = user_agent()
2018-04-21 17:11:05,249 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 93, in user_agent
2018-04-21 17:11:05,250 ! from pip._vendor import distro
2018-04-21 17:11:05,251 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/distro.py", line 1050, in <module>
2018-04-21 17:11:05,251 ! _distro = LinuxDistribution()
2018-04-21 17:11:05,252 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/distro.py", line 594, in __init__
2018-04-21 17:11:05,253 ! if include_lsb else {}
2018-04-21 17:11:05,253 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/distro.py", line 933, in _get_lsb_release_info
2018-04-21 17:11:05,254 ! raise subprocess.CalledProcessError(code, cmd, stdout)
2018-04-21 17:11:05,255 ! CalledProcessError: Command 'lsb_release -a' returned non-zero exit status 1
2018-04-21 17:11:05,256 ! Traceback (most recent call last):
2018-04-21 17:11:05,257 ! File "/usr/lib/python2.7/runpy.py", line 162, in _run_module_as_main
2018-04-21 17:11:05,257 ! "__main__", fname, loader, pkg_name)
2018-04-21 17:11:05,258 ! File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
2018-04-21 17:11:05,259 ! exec code in run_globals
2018-04-21 17:11:05,259 ! File "/home/pi/oprint/lib/python2.7/site-packages/pip/__main__.py", line 19, in <module>
2018-04-21 17:11:05,260 ! sys.exit(pip.main())
2018-04-21 17:11:05,261 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/__init__.py", line 233, in main
2018-04-21 17:11:05,261 ! return command.main(cmd_args)
2018-04-21 17:11:05,262 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/basecommand.py", line 251, in main
2018-04-21 17:11:05,263 ! timeout=min(5, options.timeout)) as session:
2018-04-21 17:11:05,263 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/basecommand.py", line 72, in _build_session
2018-04-21 17:11:05,264 ! insecure_hosts=options.trusted_hosts,
2018-04-21 17:11:05,265 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 329, in __init__
2018-04-21 17:11:05,265 ! self.headers["User-Agent"] = user_agent()
2018-04-21 17:11:05,266 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 93, in user_agent
2018-04-21 17:11:05,267 ! from pip._vendor import distro
2018-04-21 17:11:05,267 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/distro.py", line 1050, in <module>
2018-04-21 17:11:05,268 ! _distro = LinuxDistribution()
2018-04-21 17:11:05,268 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/distro.py", line 594, in __init__
2018-04-21 17:11:05,269 ! if include_lsb else {}
2018-04-21 17:11:05,270 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/distro.py", line 933, in _get_lsb_release_info
2018-04-21 17:11:05,271 ! raise subprocess.CalledProcessError(code, cmd, stdout)
2018-04-21 17:11:05,271 ! subprocess.CalledProcessError: Command 'lsb_release -a' returned non-zero exit status 1

jneilliii, 2018-04-21:
Not really sure then. I'd try restarting and start from the beginning.

maddavo, 2018-04-24:
I'm wondering if this might be a pip vs pip3 issue.  I have python 3.4 and pip3 installed separately and even though Octoprint has its own python2.7 install local to it, I wonder if my path/env variable may be interfering.  I'll change some things and see how it goes.

maddavo, 2018-04-25:
Yes, that was the problem - Python 3 was set as the default python in /usr/bin .  So somewhere something was referencing python in /usr/bin directory even though Octoprint has its own python2.7 install in /home/pi/oprint/bin .

closed 2018-04-25 by maddavo