jneilliii/OctoPrint-BedLevelVisualizer #26: Plotly graph not displaying

Mavyre, 2018-04-22:
Hello,

Plotlyjs doesn't display on my Octoprint instance, and I can't figure out how to get it working. The stored mesh data is okay, and the Bed Visualizer tab only show the Update button.
Anyone having this issue too?

jneilliii, 2018-04-22:
Typically it's because you haven't set the data collector flag and gcode settingsvcorrectly. Look at [readme](readme.md).

Mavyre, 2018-04-22:
It's all set correctly! But the thing is it displays 3 different maps and
the plug-in register all them as one... I'll send you the output of my
command (Merlin) and what data is saved by the plug-in

AikiAviator, 2018-04-23:
Hi,
I have a standard Prusa i3 Mk3, and am new to this stuff. What would be the standard GCODES for the Maps codes.

Regards,
Andrew

jneilliii, 2018-04-23:
I think you're gcode should be G81 and the data collector flag should be Measured points:

dafyddhirst, 2018-04-24:
Im having similar issues .But on a cr10S tuning a TH3d sensor. 

i get TypeError: Plotly.react is not a function

GCODE - 
M155 S30
G28
G29 T
M155 S3 

Data colector flag set to         Bilinear Leveling Grid:

Results from terminal 
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3
Recv:  0 -0.156 -0.159 -0.173 -0.162
Recv:  1 +0.028 -0.035 -0.105 -0.105
Recv:  2 +0.024 -0.010 -0.053 -0.030
Recv:  3 -0.152 -0.087 -0.089 -0.079
Recv: 
Recv: X:299.00 Y:295.00 Z:8.07 E:0.00 Count X:23920 Y:23600 Z:3196
Recv: ok

Stored Mesh Data Shows 
Stored Mesh Data:
0	-0.166	-0.166	-0.180	-0.164
1	+0.021	-0.039	-0.104	-0.103
2	+0.020	-0.013	-0.056	-0.029
3	-0.155	-0.087	-0.090	-0.077

If i click update it updates the data but still shows update button and a red box apears showing 

Bed Visualizer Error
Looks like your settings are not correct or there was an error. Please see the Readme for configuration hints.
TypeError: Plotly.react is not a function

Any help appreciated.

jneilliii, 2018-04-24:
This plotyly.react error is because of a conflict with the tempsgraph plugin, which should havean update soon to resolve.

dafyddhirst, 2018-04-24:
Yes i can confirm disabling tempsgraph resolves the issue.

jneilliii, 2018-04-24:
@dafyddhirst: There is a new release of TempsGraph published that resolves the conflict.

jneilliii, 2018-04-29:
I've pushed an 0.1.0 branch that simplifies the data collection process drastically.  Can you please try this one by installing the plugin from the URL below and let me know if it helps with your errors?

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.0.zip

closed 2018-05-01 by jneilliii