jneilliii/OctoPrint-BedLevelVisualizer #27: No displays timestamp on mouse hover

ShagoY, 2018-04-22:
Hi Jneilliii :)
It's doesn't work. I see tooltips diretly on html code but not in Chrome brower.

jneilliii, 2018-04-22:
Do you have the option to save the mesh enabled in settings?  It will only show the timestamp when that option is enabled.  It display on mouse hover over the "i" symbol on the button.

ShagoY, 2018-04-22:
Yes.

Btw i see on html code : 
<i class="icon-2 icon-info-sign" data-bind="attr: {'title': mesh_status}" title="Using saved mesh data from 22/04/2018 à 17:51:17."></i>

jneilliii, 2018-04-22:
Yeah, the code is showing it their, and I use Chrome and it's working for me when I hover the mouse over the i symbol.

[image]

jneilliii, 2018-04-22:
I didn't want the timestamp to be completely visible, but there if you wanted to see when it was run.  The title attribute is the best method I could come up with for that.

ShagoY, 2018-04-22:
Oh nevermind... "It display on mouse hover over the "i" symbol on the button."

jneilliii, 2018-04-29:
I've pushed an 0.1.0 branch that utilizes bootstrap tooltips.  Can you please try this one by installing the plugin from the URL below and let me know if it helps with not seeing the timestamp?

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.0.zip

closed 2018-05-01 by jneilliii