jneilliii/OctoPrint-BedLevelVisualizer #32: BedLevel 0.0.8 error

ytymec, 2018-04-23:
[image]
This error is occuring every time I'm clicking on the BedLevel tab.
Octoprint is on the latest version

TheBennyJ, 2018-04-23:
I'm getting the same
[image]

jneilliii, 2018-04-23:
Try pressing the update button once on the tab to retrieve the mesh data and store in memory.

TheBennyJ, 2018-04-23:
That's causing the error, after pressing the button it starts running the GCODE, after the printer displays the content for G29 T, then that message appears. Was working prior to the 0.8 update. 

sample of my G29 T

Recv: Bed Topography Report:
Recv: 
Recv: (0,6)                                           (6,6)
Recv: (28,172)                                        (172,172)
Recv:  -218.773   -213.773   -222.142   -224.626   -227.621   -235.415   -252.802
Recv: 
Recv:  -216.400   -213.127   -214.972   -214.065   -213.095   -225.686   -255.163
Recv: 
Recv:  -216.138   -201.159   -205.243   -197.428   -212.897   -218.835   -236.031
Recv: 
Recv:  -208.904   -199.042   -193.531   -200.172   -201.697   -210.386   -240.183
Recv: 
Recv:  -203.269   -200.124   -198.322   -200.614   -210.710   -218.631   -236.019
Recv: 
Recv:  -203.518   -197.943   -193.902   -205.341   -207.953   -211.908   -231.215
Recv: 
Recv:  -214.898   -196.849   -195.751   -206.614   -209.226   -214.077  [-230.953]
Recv: (28,28)                                            (172,28)
Recv: (0,0)                                             (6,0)

TheBennyJ, 2018-04-23:
look at the values, my mesh got bad values somewhere. I re-ran bed leveling and the visualizer is working correctly again. 

These values work. 

Recv: Bed Topography Report:
Recv: 
Recv: (0,6)                                           (6,6)
Recv: (28,172)                                        (172,172)
Recv:   0.703    0.795    0.899    0.772    0.744    0.671    0.540
Recv: 
Recv:   0.632    0.807    0.718    0.791    0.668    0.617    0.482
Recv: 
Recv:   0.579    0.717    0.742    0.688    0.606    0.579    0.424
Recv: 
Recv:   0.475    0.502    0.633  [ 0.546]   0.505    0.538    0.367
Recv: 
Recv:   0.451    0.370    0.372    0.362    0.333    0.409    0.241
Recv: 
Recv:   0.305    0.323    0.256    0.274    0.298    0.181    0.192
Recv: 
Recv:   0.166    0.177    0.127    0.131    0.177    0.094    0.068
Recv: (28,28)                                            (172,28)
Recv: (0,0)

ytymec, 2018-04-24:
Still the same issue, adding that after pressing update, the screen stay on "retrieving current mesh"...
Additional information: I'm on old Marlin version : 1.1.0-RC6

jneilliii, 2018-04-24:
@ytymec: what's the output of manually sending the command `G29 T`?

ytymec, 2018-04-25:
[...]
Send: G29 T
Recv: G29 Auto Bed Leveling
[...]
Recv: Eqn coefficients: a: 0.00198333 b: 0.00386000 d: -0.97882490
Recv: 
Recv: Bed Height Topography:
Recv:    +--- BACK --+
Recv:    |           |
Recv:  L |    (+)    | R
Recv:  E |           | I
Recv:  F | (-) N (+) | G
Recv:  T |           | H
Recv:    |    (-)    | T
Recv:    |           |
Recv:    O-- FRONT --+
Recv:  (0,0)
Recv:  +0.37375 +0.20525
Recv:  -0.64150 +0.06250
Recv: 
Recv: 
Recv: 
Recv: Bed Level Correction Matrix:
Recv: +0.999998 +0.000000 +0.001983
Recv: -0.000008 +0.999993 +0.003860
Recv: -0.001983 -0.003860 +0.999991

jneilliii, 2018-04-25:
Yus should set your data collector flag to `Bed Level Correction Matrix:`

ytymec, 2018-04-25:
It works with the latest version of Marlin 🥁

closed 2018-04-25 by ytymec