jneilliii/OctoPrint-BedLevelVisualizer #39: I can't seem to get the plug in running?

Tonydangelo, 2018-04-24:
When I click on the plugin I get this error, do i need to install Plotly.js? if so do I install it on the raspberry pi? or the PC I am using to view? or is it a completely different issue? I am new to this but reading as much as I can find.
Thank you very much.
Tony
Bed Visualizer Error
Looks like your settings are not correct or there was an error. Please see the Readme for configuration hints.

TypeError: s[0] is undefined

Tonydangelo, 2018-04-25:
Thank you very much for your help. I have done an update on the pi, and I still get the error “type error: s[0} is undefined. I am unsure what to put in under correct data collector flag in the plug-in settings in the plug-in settings. When I run the update on the plug-in now the printer runs its auto leveling process and displays “Please wait, retrieving current mesh.” 
Thank you very much this is all new to me, and I do appreciate you help very much.
Tony

Tonydangelo, 2018-04-25:
Sir This is what I get on the g29 -t
the printer does run through it leveling process.

Thank you very much
Tony

Send: G29 -TRecv:  T:25.62 /0.00 B:24.81 /0.00 @:0 B@:0Recv:  T:25.56 /0.00 B:24.69 /0.00 @:0 B@:0Recv:  T:25.66 /0.00 B:24.62 /0.00 @:0 B@:0Recv:  T:25.69 /0.00 B:24.75 /0.00 @:0 B@:0Recv:  T:25.66 /0.00 B:24.81 /0.00 @:0 B@:0Recv:  T:25.56 /0.00 B:24.75 /0.00 @:0 B@:0Recv:  T:25.59 /0.00 B:24.84 /0.00 @:0 B@:0Recv:  T:25.53 /0.00 B:24.87 /0.00 @:0 B@:0Recv:  T:25.66 /0.00 B:24.84 /0.00 @:0 B@:0Recv:  T:25.66 /0.00 B:24.78 /0.00 @:0 B@:0Recv:  T:25.59 /0.00 B:24.78 /0.00 @:0 B@:0Recv:  T:25.59 /0.00 B:24.69 /0.00 @:0 B@:0Recv:  T:25.59 /0.00 B:24.78 /0.00 @:0 B@:0Recv:  T:25.47 /0.00 B:24.84 /0.00 @:0 B@:0Recv:  T:25.56 /0.00 B:24.75 /0.00 @:0 B@:0Recv:  T:25.53 /0.00 B:24.75 /0.00 @:0 B@:0Recv:  T:25.56 /0.00 B:24.78 /0.00 @:0 B@:0Recv:  T:25.53 /0.00 B:24.84 /0.00 @:0 B@:0Recv:  T:25.69 /0.00 B:24.78 /0.00 @:0 B@:0Recv:  T:25.59 /0.00 B:24.84 /0.00 @:0 B@:0Recv:  T:25.59 /0.00 B:24.94 /0.00 @:0 B@:0Recv:  T:25.62 /0.00 B:24.94 /0.00 @:0 B@:0Recv:  T:25.59 /0.00 B:24.87 /0.00 @:0 B@:0Recv:  T:25.69 /0.00 B:24.84 /0.00 @:0 B@:0Recv: Bilinear Leveling Grid:Recv:       0      1      2Recv:  0 -0.067 -0.152 -0.132Recv:  1 +0.090 -0.002 -0.048Recv:  2 +0.105 +0.038 +0.085Recv: Recv: X:214.00 Y:214.00 Z:10.81 E:22114.33 Count X:21400 Y:21400 Z:4360Recv: okRecv:  T:25.59 /0.00 B:24.84 /0.00 @:0 B@:0Recv:  T:25.56 /0.00 B:24.78 /0.00 @:0 B@:0Recv:  T:25.53 /0.00 B:24.81 /0.00 @:0 B@:0Recv:  T:25.59 /0.00 B:24.84 /0.00 @:0 B@:0Recv:  T:25.72 /0.00 B:24.91 /0.00 @:0 B@:0Recv:  T:25.56 /0.00 B:24.84 /0.00 @:0 B@:0Recv:  T:25.47 /0.00 B:24.78 /0.00 @:0 B@:0Recv:  T:25.53 /0.00 B:24.84 /0.00 @:0 B@:0Recv:  T:25.59 /0.00 B:24.91 /0.00 @:0 B@:0Recv:  T:25.53 /0.00 B:24.94 /0.00 @:0 B@:0Recv:  T:25.56 /0.00 B:24.87 /0.00 @:0 B@:0Recv:  T:25.66 /0.00 B:24.78 /0.00 @:0 B@:0

jneilliii, 2018-04-25:
Ok, in your case you are running into an issue with the temperature reports interfering with the mesh data.  Do you have Marlin's auto temperature reporting?

jneilliii, 2018-04-25:
Change your gcode script to the following.

```
M155 S30
G28
G29 T
M155 S3
```

Tonydangelo, 2018-04-25:
Sorry it took so long to reply printer was down for up grades. 
In config_adv.h auto temp is cometed out. 
See bellow

//#define AUTOTEMP
#if ENABLED(AUTOTEMP)
  #define AUTOTEMP_OLDWEIGHT 0.98
#endif

In the setup page for bed visualize I had these under gcoad commands. 
See bellow

M155 S30
G29 T
M155 S3

I added the G28 command after the M155 S30 as direacted but I have gotten the same result. 

Thank You
Tony

jneilliii, 2018-04-25:
Sorry about that, you are correct. If it's commented out it's not enabled. You should use

```
G28
G29 T
```

and in the data collector flag setting enter

```
Bilinear Leveling Grid:
```

Tonydangelo, 2018-04-25:
Wow! I am getting data! I still get thr same error message but who cares I am getting a map. with regard to leveling I am looking to get te chart as close to all zeros? ie flat?
Thank man!