jneilliii/OctoPrint-BedLevelVisualizer #40: Seems to be displaying values that are flipped on the Y-axis

Zer0t3ch, 2018-04-25:
I'm on my [personal fork](https://github.com/Zer0t3ch/Marlin) of the Marlin bugfix-2.0.x branch using UBL. The default branch of my repo is what I'm compiling right now, so you can have a look there if you need to see my Marlin configuration headers.

[image]
[image]

jneilliii, 2018-04-25:
Thanks for this information.  No-one has reported on the axis order yet.  I was concerned it would be flipped for some. I'll review for my next set of revisions related to #28.

Zer0t3ch, 2018-04-26:
@jneilliii 

Note: the output format seems to have changed in the 2.0.x branch, recently. Here's my current output, for reference:

```
Recv: Bed Topography Report:Recv: Recv:     ( 50,170)              (170,170)Recv:         0       1       2       3Recv:  3 |  0.000   0.000   0.000   0.000Recv:    |Recv:  2 |  0.000   0.000   0.000   0.000Recv:    |Recv:  1 |  0.000   0.000   0.000   0.000Recv:    |Recv:  0 |  0.000   0.000 [ 0.000]  0.000Recv:         0       1       2       3Recv:     ( 50, 50)              (170, 50)
```

jneilliii, 2018-04-26:
I just pushed a commit to the 0.0.9 branch with the Flip X/Y axis options.  I also updated the collector regular expression, which will probably break delta printer output, but would work for the 2.0.x report that you posted before in theory.  You can test by installing with the below url in plugin manager.  Make sure to CTRL+F5 afterwards just to make sure the page is refreshed.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.0.9.zip

Zer0t3ch, 2018-04-27:
Seems to be working as expected. (for both the tabbed-CSV output and flipping) Thank you.

Do you intend to debug/fix the flipped mesh data issue directly, or is the manual flipping configuration your long-term solution? (wasn't sure if I should close the issue or not)

jneilliii, 2018-04-27:
I think this will be the long term solution.  There are too many variables when dealing with the different firmware and how they report the mesh data.

closed 2018-04-27 by jneilliii