jneilliii/OctoPrint-BedLevelVisualizer #41: Erroneous values in plot

5ft24dave, 2018-04-25:
After obtaining data from my 10x10 pattern, the plot that comes up shows X0, Y9 to have a Z value of 9 as well.  The actual matrix data isn't showing anything near that value in Z.

Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4      5      6      7      8      9
Recv:  0 +0.582 +0.472 +0.382 +0.292 +0.227 +0.152 +0.087 +0.025 -0.025 -0.083
Recv:  1 +0.552 +0.427 +0.307 +0.225 +0.155 +0.082 +0.012 -0.053 -0.098 -0.128
Recv:  2 +0.482 +0.377 +0.262 +0.185 +0.115 +0.052 -0.013 -0.068 -0.120 -0.173
Recv:  3 +0.445 +0.315 +0.192 +0.112 +0.042 -0.023 -0.085 -0.133 -0.173 -0.193
Recv:  4 +0.410 +0.300 +0.187 +0.115 +0.042 -0.023 -0.080 -0.118 -0.155 -0.205
Recv:  5 +0.400 +0.270 +0.157 +0.067 -0.003 -0.063 -0.120 -0.158 -0.185 -0.208
Recv:  6 +0.420 +0.295 +0.190 +0.090 +0.032 -0.025 -0.073 -0.108 -0.143 -0.198
Recv:  7 +0.435 +0.300 +0.172 +0.077 +0.010 -0.040 -0.085 -0.115 -0.153 -0.183
Recv:  8 +0.465 +0.357 +0.237 +0.150 +0.075 +0.025 -0.018 -0.058 -0.100 -0.170
Recv:  9 +0.490 +0.305 +0.260 +0.140 +0.072 +0.045 -0.013 -0.068 -0.105 -0.138

[image]

jneilliii, 2018-04-25:
I'll have to look into this one.  I believe I know the reason, it's because the 9 didn't get stripped off the last row during the collection process.  To verify, it you look in the plugin settings at the stored mesh, the last row will have a 9 in it?

5ft24dave, 2018-04-26:
0	+0.582	+0.472	+0.382	+0.292	+0.227	+0.152	+0.087	+0.025	-0.025	-0.083
1	+0.552	+0.427	+0.307	+0.225	+0.155	+0.082	+0.012	-0.053	-0.098	-0.128
2	+0.482	+0.377	+0.262	+0.185	+0.115	+0.052	-0.013	-0.068	-0.120	-0.173
3	+0.445	+0.315	+0.192	+0.112	+0.042	-0.023	-0.085	-0.133	-0.173	-0.193
4	+0.410	+0.300	+0.187	+0.115	+0.042	-0.023	-0.080	-0.118	-0.155	-0.205
5	+0.400	+0.270	+0.157	+0.067	-0.003	-0.063	-0.120	-0.158	-0.185	-0.208
6	+0.420	+0.295	+0.190	+0.090	+0.032	-0.025	-0.073	-0.108	-0.143	-0.198
7	+0.435	+0.300	+0.172	+0.077	+0.010	-0.040	-0.085	-0.115	-0.153	-0.183
8	+0.465	+0.357	+0.237	+0.150	+0.075	+0.025	-0.018	-0.058	-0.100	-0.170
9	+0.490	+0.305	+0.260	+0.140	+0.072	+0.045	-0.013	-0.068	-0.105	-0.138

Yep

jneilliii, 2018-04-29:
Can you try the 0.1.0 branch I just pushed and let me know if results are the same?  Use the url below in plugin manager to install.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.0.zip

5ft24dave, 2018-04-29:
You got it.  Probably won't be until late tonight or tomorrow morning as I have a big print running right now

⁣Sent from BlueMail ​

5ft24dave, 2018-04-30:
> archive/0.1.0.zip

Works great!!!

jneilliii, 2018-04-30:
Thanks. I've received reports that this version doesn't update after the first data pull, so subsequent clicks of the update button doesn't refresh the data. Was goingto look into that tonight.

5ft24dave, 2018-04-30:
I was just going to email you that LOL. And I'm using a 10x10 matrix, and after the first 2 rows, updating goes away and the graph pops back up

⁣Sent from BlueMail ​

jneilliii, 2018-04-30:
Try this...if you have multiple commands in your GCODE setting, place the text `@BEDLEVELVISUALIZER` just above the `G29` command...like this.

[image]

What is happening is the collector stops when it receives an ok command, and if that flag isn't in the gcode you enter the ok is coming earlier than expected.

5ft24dave, 2018-04-30:
so far, so good... 5 rows in and hasn't brought up the graph yet...  I'll
let you know the final result here in a few

5ft24dave, 2018-04-30:
Working perfectly!

5ft24dave, 2018-04-30:
new issue!!!
every time it runs now, it adds to the stored data with the new data...
doesn't clear the old...

Bed Visualizer Settings

   - General <http://10.0.0.195:8188/#bedlevelvisualizer_general>
   - Stored Data <http://10.0.0.195:8188/#bedlevelvisualizer_stored_data>

Stored X Positions:

0	24	49	73	98	122	147	171	196	220

Stored Y Positions:

0	6	11	17	23	28	34	39	45	51	56	62	68	73	79	85	90	96	102	107	113	118	124	130	135	141	147	152	158	164	169	175	181	186	192	197	203	209	214	220

Stored Mesh Data:

+0.457	+0.287	+0.347	+0.305	+0.277	+0.257	+0.237	+0.225	+0.217	+0.217
+0.420	+0.345	+0.265	+0.225	+0.197	+0.170	+0.142	+0.127	+0.120	+0.137
+0.345	+0.267	+0.195	+0.165	+0.140	+0.117	+0.102	+0.087	+0.080	+0.072
+0.290	+0.197	+0.112	+0.075	+0.050	+0.022	+0.002	+0.005	+0.005	+0.027
+0.237	+0.160	+0.095	+0.062	+0.027	+0.005	-0.010	-0.008	-0.000	-0.005
+0.217	+0.125	+0.055	-0.000	-0.028	-0.048	-0.065	-0.053	-0.048	-0.025
+0.217	+0.137	+0.067	+0.012	-0.013	-0.020	-0.025	-0.028	-0.023	-0.030
+0.220	+0.127	+0.040	-0.015	-0.038	-0.055	-0.065	-0.053	-0.048	-0.038
+0.225	+0.157	+0.085	+0.040	+0.007	-0.010	-0.010	-0.010	-0.015	-0.035
+0.232	+0.152	+0.095	+0.012	-0.020	-0.005	-0.020	-0.030	-0.040	-0.028
+0.385	+0.202	+0.297	+0.267	+0.247	+0.227	+0.215	+0.205	+0.205	+0.215
+0.355	+0.285	+0.215	+0.180	+0.160	+0.140	+0.120	+0.102	+0.110	+0.130
+0.292	+0.227	+0.162	+0.137	+0.117	+0.102	+0.095	+0.082	+0.080	+0.072
+0.240	+0.165	+0.085	+0.055	+0.037	+0.015	+0.002	+0.012	+0.020	+0.045
+0.212	+0.145	+0.082	+0.060	+0.035	+0.017	+0.007	+0.015	+0.025	+0.020
+0.195	+0.117	+0.050	+0.002	-0.020	-0.035	-0.043	-0.038	-0.015	+0.010
+0.210	+0.135	+0.070	+0.020	+0.012	-0.005	-0.008	+0.010	+0.015	+0.007
+0.217	+0.132	+0.050	-0.000	-0.018	-0.028	-0.028	-0.015	-0.005	+0.017
+0.227	+0.175	+0.107	+0.065	+0.040	+0.030	+0.030	+0.040	+0.042	+0.020
+0.245	+0.177	+0.122	+0.047	+0.017	+0.040	+0.030	+0.015	+0.025	+0.037
+0.487	+0.282	+0.382	+0.327	+0.300	+0.267	+0.235	+0.217	+0.200	+0.185
+0.450	+0.370	+0.285	+0.237	+0.205	+0.170	+0.140	+0.112	+0.102	+0.110
+0.372	+0.300	+0.225	+0.190	+0.155	+0.127	+0.095	+0.082	+0.070	+0.045
+0.315	+0.230	+0.140	+0.092	+0.065	+0.030	+0.002	-0.010	-0.008	+0.002
+0.270	+0.200	+0.122	+0.085	+0.047	+0.017	-0.010	-0.010	-0.010	-0.028
+0.245	+0.157	+0.080	+0.020	-0.015	-0.043	-0.068	-0.068	-0.060	-0.048
+0.245	+0.162	+0.087	+0.030	-0.000	-0.020	-0.035	-0.038	-0.043	-0.063
+0.242	+0.152	+0.062	-0.003	-0.033	-0.055	-0.068	-0.065	-0.070	-0.065
+0.247	+0.180	+0.102	+0.052	+0.015	-0.010	-0.018	-0.025	-0.038	-0.075
+0.250	+0.172	+0.107	+0.022	-0.015	-0.005	-0.030	-0.053	-0.063	-0.060
+0.235	+0.145	+0.197	+0.187	+0.190	+0.195	+0.202	+0.210	+0.235	+0.262
+0.222	+0.177	+0.133	+0.120	+0.125	+0.130	+0.133	+0.137	+0.163	+0.208
+0.167	+0.133	+0.093	+0.095	+0.097	+0.105	+0.110	+0.130	+0.147	+0.165
+0.130	+0.080	+0.030	+0.020	+0.028	+0.033	+0.045	+0.060	+0.100	+0.145
+0.110	+0.075	+0.040	+0.038	+0.038	+0.043	+0.055	+0.082	+0.120	+0.135
+0.110	+0.058	+0.015	-0.013	-0.008	+0.002	+0.020	+0.055	+0.090	+0.137
+0.135	+0.087	+0.055	+0.025	+0.035	+0.053	+0.065	+0.103	+0.135	+0.147
+0.150	+0.097	+0.043	+0.015	+0.023	+0.038	+0.060	+0.100	+0.130	+0.163
+0.185	+0.153	+0.113	+0.097	+0.090	+0.100	+0.133	+0.157	+0.182	+0.185
+0.215	+0.167	+0.137	+0.090	+0.082	+0.133	+0.140	+0.153	+0.180	+0.212

 About OctoPrint

jneilliii, 2018-04-30:
Thanks.  I probably forgot to clear the previous value and it's just appending. I'm looking into some other little glitches now and will push an update shortly to the 0.1.0 branch.

5ft24dave, 2018-04-30:
Cool

⁣Sent from BlueMail ​

jneilliii, 2018-04-30:
This issue is fixed now in 0.1.0.  Reinstall and ctrl+f5 to reload cache from server and it should work fine now.  There is still a weird thing I've run into related to switching tabs, updating, and going into settings and nothing be there.  Once I've nailed that down, and hear back from the delta printer guys about the data collection working for them I'll release this version.

closed 2018-04-30 by jneilliii