jneilliii/OctoPrint-BedLevelVisualizer #42: Printrbot simple metal settings

MBarrows20, 2018-04-26:
I cant seem to find the right settings for my Printrbot Simple Metal printer. I've included screenshots of my settings and console output. I've been able to find the right settings for the Prusa i3 MK3 with the help of these issue posts so I'm hoping to have some luck with the Printrbot. I've included screenshots of the settings I'm currently using (although I've tried all of the data collector flag options). Thanks in advance! 

[image]
[image]

jneilliii, 2018-04-26:
Unfortunately, the plugin does not currently know how to interpret this format of report.  What do you get if you just send the command `G29 T`? Do you know what firmware the Printrbot Simple Metal uses?

MBarrows20, 2018-04-26:
Hi @jneilliii, thanks for your quick response. The `G29 T` command yields the same output as my screenshot above. If I run the `M115` command to print firmware version, I get the following: 

`M115Recv: FIRMWARE_NAME:Marlin V1; Sprinter/grbl mashup for gen6 FIRMWARE_URL:http://printrbot.com PROTOCOL_VERSION:1.0 MACHINE_TYPE:Printrbot EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000`

I love this addon for the MK3 and look forward to seeing it implemented on the Printrbot :)

MBarrows20, 2018-04-26:
`M420 V` and `M420` just returns an `Recv: ok` result. I've tried running `G29 V4` as well and that gives the same output as my original screenshot. If it helps, I'm operating with the Marlin list of GCodes [here.](http://reprap.org/wiki/G-code#M420:_Leveling_On.2FOff.2FFade_.28Marlin.29)

jslay88, 2018-05-06:
@MBarrows20 I'd suggest updating the firmware on your printer to 1.1.8

See here for more info on your printer.
http://www.printrbottalk.com/forum/viewtopic.php?f=115&t=11462

MBarrows20, 2018-06-04:
Unfortunately I've left the organization with this printer before I was able to try the firmware update. I hope this thread will help someone in the future but feel free to mark the issue as closed if you wish.

closed 2018-06-05 by jneilliii