jneilliii/OctoPrint-BedLevelVisualizer #44: Can't get the plugin to work

ShinSnap, 2018-04-26:
I receive an error when I first open the Bed Visualizer plugin.

"TypeError: Cannot read property 'length' of undefined"

[image]

Data Collection Flag: 
`Measured points:`

Gcode Commands:
`G28
G29 T`

Save Mesh
`Checked`

Check I click update. All I get is, but it doesn't do anything
`Please wait, retrieving current mesh.`

When I check the logs G28 and G29 T responds with `OK` almost immediately.

jneilliii, 2018-04-26:
Please share a screenshot of your settings and the output of manually sending command `G29 T`

ShinSnap, 2018-04-26:
[image]

[image]

onlyhuman91, 2018-04-27:
I am also seeing this same error message now, but I wasn't last week. Was there an update recently?

closed 2018-05-02 by jneilliii