jneilliii/OctoPrint-BedLevelVisualizer #45: TypeError: Cannot read property 'length' of undefined

rrauenza, 2018-04-26:
I've just installed this and get this error:

> TypeError: Cannot read property 'length' of undefined

in this box:

> Bed Visualizer Error
> Looks like your settings are not correct or there was an error. Please see the Readme for configuration hints.

I have the tag set to:  Bed Topography Report for CSV:
I have Save Mesh and Prusa Firmware both checked.
I have commands set to

```
G28
G29 T
```

Bed Visualizer (0.0.8)
OctoPrint: 1.3.8 

...I'm not sure where the plugin logs, so I can't get a stack trace.

jneilliii, 2018-04-26:
What firmware? If your collector is `Bed Topography Report for CSV:` your gcode should be `G29 T1` I think. Otherwise, change your collector flag to the text displayed just prior to mesh data when running `G29 T`

jneilliii, 2018-04-26:
If you're using Prusa firmware your gcode needs to be

```
G80
G81
```

closed 2018-04-26 by jneilliii

rrauenza, 2018-04-26:
Yes, I've just fixed it -- I had to change it to G28, G81 and change the tag to 'Measured points:'

It's a Prusa i3 mk3

....I was confused because I didn't expect that error code so quickly.  I would have expected it to have an error after the bed levelling was done.

rrauenza, 2018-04-26:
I think there's room to improve the error reporting here ... it's not clear at all how that error results from the incorrect tag.  I got it immediately when clicking on the UI menu without any delay from the G28 command (G28 does a G80 as well).

jneilliii, 2018-04-27:
I figured out where the error was being triggered and just updated the 0.0.8 version.  If you uninstall and reinstall you should not see these pop-up errors anymore.