jneilliii/OctoPrint-BedLevelVisualizer #46: Cant plot from recived data

Swartor, 2018-04-28:
I racive this code from G29 T
`Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv: (0,4)                           (4,4)
Recv: (15,185)                        (169,185)
Recv:  -0.239 T:24.96 /0.00 B:26.65 /0.00 @:0 B@:0
Recv:    -0.137   -0.052   -0.027    0.006
Recv: 
Recv:  -0.222   -0.155   -0.056   -0.054   -0.027
Recv: 
Recv:  -0.220   -0.156   -0.074  [-0.068]  -0.054
Recv: 
Recv:  -0.211   -0.168   -0.093   -0.092   -0.087
Recv: 
Recv:  -0.207   -0.157   -0.102   -0.112   -0.100
Recv: (15,15)                            (169,15)
Recv: (0,0)                             (4,0)
Recv: ok`

It plots with bad spot  [-0.068 dont get the right bracket
[image]

when i manualy delete the left bracket and live it without it plot OK
[image]

jneilliii, 2018-04-28:
Thanks for the data.  I just pushed updates that should resolve this for you.  Please uninstall/reinstall the plugin and let me know if that worked for you.

Swartor, 2018-04-28:
all work pok now thx and keep the good work

closed 2018-04-28 by jneilliii