jneilliii/OctoPrint-BedLevelVisualizer #50: Unknown in install popup

OutsourcedGuru, 2018-05-02:
Just installed this today (likely 0.1.0) and saw this in the octoprint.log although it may be nothing:

```
2018-05-02 13:43:21,550 - py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py:1643: UserWarning: No PKG-INFO found for package: octoprint_bedlevelvisualizer
```

...as accompanied by this popup during the install:

[image]

I really don't know much about egg-info and packaging, unfortunately.

jneilliii, 2018-05-02:
I've typically seen this happen during reinstall and when the code was bad.  Check your octoprint.log for any traceback errors please.

OutsourcedGuru, 2018-05-11:
Just found out that it's not you, it's the Pip v10 that's throwing a different metadata format which OctoPrint now can't read. So it has nothing to do with any of your plugins across the board.

closed 2018-05-11 by OutsourcedGuru