jneilliii/OctoPrint-BedLevelVisualizer #51: Bed level graphic wont display in Pi 7" screen

mritchie1, 2018-05-07:
Hi Guys,  
I've installed a BLTouch probe with Marlin 1.1.8 and Octoprint 1.3.8.  I've calibrated the BLTouch and set the Z offset, and all is working ok.  

When I connect to Octoprint from the my PC runing Windows10 and Chrome browser, the Bed Visualizer works fine, completes and I see the graphical display that represents/shows the level of the print bed surface.  

On my Pi connected to the printer I have a 7" touch screen and TouchUI running.  When I do a bed visualization in the Pi via the 7" touch screen, it completes the probing and then displays an error and does not show the graphical (picture) display that represents/shows the level of the print bed surface.  

The error I receive on the Pi 7" screen is:
Bed Visualizer Error
Looks like your settings are not correct or there was an error. Please see the readme for config hints.
TypeError: Cannot read property 'canvas' of undefined

Behind the error message is another error on what looks like the white screen where it was getting ready to display the  graphical display that represents/shows the level of the print bed surface:
Webgl is not supported by your browser - visit http://get webgl.org for more info
When you visit this URL it does a test and tells me that the browser seems to support WebGL and it is disabled or unavailable.

Not sure how to fault find this issue and where to start.  Any info or tips would be greatly appreciated.

Many thanks

Mark R

mritchie1, 2018-05-07:
I found that I could check the browser version via a telnet connection.
 chromium-browser --version
Chromium 56.0.2924.84 Built on Ubuntu 14.04, running on Raspbian 8.0
Guess this could be the browser version that TouchUI is running in.

jneilliii, 2018-05-07:
That appears to be a browser incompatibility issue. There's nothing I can do about that. I do know the visualizer works with touchui from my phone.

closed 2018-05-07 by jneilliii