jneilliii/OctoPrint-BedLevelVisualizer #52: Stuck on Please wait, retrieving current mesh.

modsbyus, 2018-05-12:
I have a Robo R2. I installed this plugin.. I love it! It worked fine the first three times I used it. Now it just gets stuck on Please wait, retrieving current mesh. If I refresh the screen (CTL+F5) and go to the bed visualizer tab, it displays the last successful plot data. When I check the terminal history I can see that it is running and getting new points. Its just not displaying the new plot.
From config.yaml
 bedlevelvisualizer:
    command: 'G28

      M155 S30

      @BEDLEVELVISUALIZER

      G29

      M155 S3'
    mesh_timestamp: 5/11/2018, 7:45:21 PM
    stored_mesh:
    - - '+0.137'
      - '+0.238'
      - '+0.639'
    - - '+0.304'
      - '-0.024'
      - '+0.045'
    - - '+0.063'
      - '+0.498'
      - '+0.276'
    stored_mesh_x:
    - 0
    - 98
    - 196
    stored_mesh_y:
    - 0
    - 98
    - 196
    stored_mesh_z_height: 250

From Terminal Output

Send: G28
Send: M155 S30
Send: @BEDLEVELVISUALIZER
Send: G29
Recv:  T:25.06 /0.00 B:23.97 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:25.03 /0.00 B:23.91 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:25.03 /0.00 B:24.03 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:25.06 /0.00 B:23.97 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:25.11 /0.00 B:24.09 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:24.92 /0.00 B:24.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:25.03 /0.00 B:24.03 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:26.28 /0.00 B:24.84 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: X:0.00 Y:197.00 Z:255.95 E:0.00 Count X:0 Y:15760 Z:204760
Recv: ok
Send: M155 S3
Recv: ok
Recv: echo:Unknown command: "@BEDLEVELVISUALIZER"
Recv: ok
Recv: echo:busy: processing
Recv: echo:busy: processing
Send: M105
Recv: echo:busy: processing
Recv: echo:busy: processing
Send: M105
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:25.43 /0.00 B:24.06 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 +11.154 +10.972 +11.796
Recv:  1 +11.334 +11.307 +11.772
Recv:  2 +11.451 +11.399 +11.870
Recv: 
Recv: X:184.00 Y:156.00 Z:10.94 E:0.00 Count X:14720 Y:12480 Z:18208
Recv: ok
Send: M105
Recv: ok

Any help would be very appreciated

jneilliii, 2018-05-13:
Try making `@bedlevelvisualizer` all caps.

modsbyus, 2018-05-13:
I think its lowercase because after adding it I checked the option to save mesh.  Because it is all uppercase in the gcode commands window. By unchecking the save mesh option it shows up as all capital letters in the terminal output without making any other changes and it still gets stuck on Please wait, retrieving current mesh.

jneilliii, 2018-05-13:
The results should only be stored in config.yaml if the option is checked to save mesh. Not sure why the command is getting lowercased when it's being sent to the machine, but I think that is the issue.

modsbyus, 2018-05-13:
It only gets lowercased when the save mesh option is not checked. but the issue of being stuck is present weather it is upper case or not. and weather save mesh is checked or not. I have uninstalled, rebooted, then reinstalled with no change.

jneilliii, 2018-05-14:
This is very similar to another user's issue where it would only be an issue after first powering on his printer that any command sent after a G28 would show as unknown. The problem I'm seeing in the terminal output is that the commands are being sent, but the plugin is receiving an ok prior to the mesh data.  

The way the plugin works is it detects sending the command `@BEDLEVELVISUALIZER` to the printer and then starts watching the commands coming back (Recv: lines).  It will collect the lines that match a specific pattern which I can tell yours will work fine.  And continues to do that until it receives an ok command back from the printer.  

In the terminal output you are showing I'm seeing M105 commands, which means you probably don't have auto temperature reporting and should take out the M155 commands from the script.  Try the following in your gcode script after power cycling your printer and pi and let me know if the results are any different.

```
G28
@BEDLEVELVISUALIZER
G29 T
```

modsbyus, 2018-05-16:
That did the trick. Thank you!

closed 2018-05-16 by jneilliii

KalCorp, 2018-06-01:
I am getting the same error.  But my G29 output is different.

Its a P802MA

Send: G28
Recv: ok
Send: G29 T1
Recv: Bed x: 20.00 y: 20.00 z: 1.80
Recv: Bed x: 90.00 y: 20.00 z: 1.73
Recv: Bed x: 160.00 y: 20.00 z: 1.22
Recv: Bed x: 160.00 y: 90.00 z: 1.20
Recv: Bed x: 90.00 y: 90.00 z: 1.73
Recv: Bed x: 20.00 y: 90.00 z: 1.77
Recv: Bed x: 20.00 y: 160.00 z: 1.94
Recv: Bed x: 90.00 y: 160.00 z: 1.93
Recv: Bed x: 160.00 y: 160.00 z: 1.36
Recv: Eqn coefficients: a: -0.00 b: 0.00 d: 1.90
Recv: planeNormal x: 0.00 y: -0.00 z: 1.00
Recv: ok
Recv: echo:endstops hit:  Z:1.37