jneilliii/OctoPrint-BedLevelVisualizer #53: Compatibility For Prusa i3 MK3

PepperManP, 2018-05-17:
hi I wanted to know if this was compatible for Prusa i3 MK3 since mine and the other owners I know are not working

jneilliii, 2018-05-17:
It should work for you as long as you use the correct sequence of gcode commands. My understanding is that the commands necessary for you are

```
G80
@BEDLEVELVISUALIZER
G81
```

let me knowif that doesn't work and provide as much detail as possible including terminal output, check for errors in octoprint.log, etc.

ShakataGaNai, 2018-05-28:
I just finished building my Mk3 and tried this out. I can confirm the code above works. Once executed it runs a calibration pass on all 9 points and my octoprint happily spit out a graph.

[image]

jneilliii, 2018-05-28:
Thanks for confirmation.

closed 2018-05-28 by jneilliii