jneilliii/OctoPrint-BedLevelVisualizer #54: Stuck on Please wait, retrieving current mesh on P802MA

KalCorp, 2018-06-01:
Stuck on Please wait, retrieving current mesh.

But my G29 output is different.

Its a P802MA 

In:
G28
@BEDLEVELVISUALIZER
G29 T

Output:
Send: G28
Recv: ok
Send: G29 T1
Recv: Bed x: 20.00 y: 20.00 z: 1.80
Recv: Bed x: 90.00 y: 20.00 z: 1.73
Recv: Bed x: 160.00 y: 20.00 z: 1.22
Recv: Bed x: 160.00 y: 90.00 z: 1.20
Recv: Bed x: 90.00 y: 90.00 z: 1.73
Recv: Bed x: 20.00 y: 90.00 z: 1.77
Recv: Bed x: 20.00 y: 160.00 z: 1.94
Recv: Bed x: 90.00 y: 160.00 z: 1.93
Recv: Bed x: 160.00 y: 160.00 z: 1.36
Recv: Eqn coefficients: a: -0.00 b: 0.00 d: 1.90
Recv: planeNormal x: 0.00 y: -0.00 z: 1.00
Recv: ok
Recv: echo:endstops hit:  Z:1.37

jneilliii, 2018-06-01:
Yeah, looks like you may have smoothieware firmware.  This plugin doesn't yet support that.  Please see #22 .

jneilliii, 2018-06-01:
Could also be an older marlin firmware as in #42, where the recommendation is to upgrade firmware.

closed 2018-06-01 by jneilliii

jneilliii, 2018-07-29:
I have a test version ready for you @KalCorp to test, if you're willing.  Install using the following url in Plugin Manager and let me know how it works out.  The install will take a little time as it will need to install an additional python library, just wait out the install, it will finish.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.3.zipe

jneilliii, 2018-07-29:
Be aware that the points won't match exactly the x/y coordinates as it interprets that from the overall size of the bed rather than the inset point. It should be close enough to get an idea of tilt/warping though.

jneilliii, 2018-08-03:
@KalCorp, have you had a chance to test yet?  I'd like to release this version if it is.

KalCorp, 2018-08-05:
Looks good!  Thanks!