jneilliii/OctoPrint-BedLevelVisualizer #55: WebGL error on iPad Safari

mark-psl, 2018-06-02:
Hi,

I’m getting the error displayed in the attached file.  When I go to the WebGL site and test the browser I confirm it supports WebGL (I see the spinning cube).  Any thoughts on what might be causing this?
[image]

jneilliii, 2018-06-02:
I just pushed a new 0.1.1 branch that potentially could resolve this.  If you don't mind testing by installing using the following url in plugin manager and let me know that would be great.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.1.zip

mark-psl, 2018-06-02:
That did it.  Thanks!

BTW, your plug-in is a life saver.  Helps me level the bed to get my EZABL working correctly.  

Mark G.

PR #56 (2018-06-02): 0.1.1
Patch plotyly js library to recognize ipad safari as a webgl device to resolve #55
closed 2018-06-02 by jneilliii