jneilliii/OctoPrint-BedLevelVisualizer #57: Bed Visualizer Error

ipproekt, 2018-06-04:
"TypeError: Cannot read property 'canvas' of undefined Looks like your settings are not correct or there was an error. "

Settings:
G29 T

Marlin 1.1.8 UBL

Send: G29 T
Recv: echo:������������ XYZ ������������
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv: (0,2)           (2,2)
Recv: (50,250)        (250,250)
Recv:  -0.053   -0.188   -0.268
Recv: 
Recv:   0.003   -0.128   -0.241
Recv: 
Recv: [ 0.141]  -0.145   -0.357
Recv: (50,50)            (250,50)
Recv: (0,0)             (2,0)
Recv: ok

Whats wrong?

jneilliii, 2018-06-04:
Which browser are you using?

jneilliii, 2018-06-05:
Can you post your settings please?

jneilliii, 2018-06-05:
Cool, that makes sense.  I didn't even realize you could turn off webgl, thanks for sharing.

closed 2018-06-05 by jneilliii