jneilliii/OctoPrint-BedLevelVisualizer #58: Prusa G81 Y-axis inverted

aderusha, 2018-06-04:
**Printer:** Prusa Mk2.5,
**Prusa Firmware:** [v3.2.1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.2.1)
**OctoPrint:** [v1.3.8](https://github.com/foosel/OctoPrint/releases/tag/1.3.8) on Raspbian
**OctoPrint-BedLevelVisualizer:** [v0.1.1](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/releases/tag/0.1.1)

## Description

OctoPrint-BedLevelVisualizer appears to be inverting the Y-axis from the Prusa `G81` output.  See examples below for details.

## G81 Output

```
0.14167	0.31361	0.43278	0.49917	0.51278	0.47361	0.38167
0.17287	0.30452	0.39569	0.44639	0.45662	0.42637	0.35565
0.19148	0.29026	0.35884	0.39722	0.40541	0.38341	0.33120
0.19750	0.27083	0.32222	0.35167	0.35917	0.34472	0.30833
0.19093	0.24624	0.28584	0.30972	0.31788	0.31032	0.28704
0.17176	0.21649	0.24970	0.27139	0.28155	0.28020	0.26731
0.14000	0.18157	0.21380	0.23667	0.25019	0.25435	0.24917
```

## BedLevelVisualizer result

|     | 0       | 42      | 83      | 125     | 167     | 208     | 250     |
|-----|---------|---------|---------|---------|---------|---------|---------|
| 0   | 0.17287 | 0.30452 | 0.39569 | 0.44639 | 0.45662 | 0.42637 | 0.35565 |
| 35  | 0.19148 | 0.29026 | 0.35884 | 0.39722 | 0.40541 | 0.38341 | 0.33120 |
| 70  | 0.19750 | 0.27083 | 0.32222 | 0.35167 | 0.35917 | 0.34472 | 0.30833 |
| 105 | 0.19093 | 0.24624 | 0.28584 | 0.30972 | 0.31788 | 0.31032 | 0.28704 |
| 140 | 0.17176 | 0.21649 | 0.24970 | 0.27139 | 0.28155 | 0.28020 | 0.26731 |
| 175 | 0.14000 | 0.18157 | 0.21380 | 0.23667 | 0.25019 | 0.25435 | 0.24917 |

## Expected result

|     | 0       | 42      | 83      | 125     | 167     | 208     | 250     |
|-----|---------|---------|---------|---------|---------|---------|---------|
| 210 | 0.17287 | 0.30452 | 0.39569 | 0.44639 | 0.45662 | 0.42637 | 0.35565 |
| 175 | 0.19148 | 0.29026 | 0.35884 | 0.39722 | 0.40541 | 0.38341 | 0.33120 |
| 140 | 0.19750 | 0.27083 | 0.32222 | 0.35167 | 0.35917 | 0.34472 | 0.30833 |
| 105 | 0.19093 | 0.24624 | 0.28584 | 0.30972 | 0.31788 | 0.31032 | 0.28704 |
| 70  | 0.17176 | 0.21649 | 0.24970 | 0.27139 | 0.28155 | 0.28020 | 0.26731 |
| 35  | 0.14000 | 0.18157 | 0.21380 | 0.23667 | 0.25019 | 0.25435 | 0.24917 |

jneilliii, 2018-06-05:
Did you try settings the option to flip the Y axis?

[image]

aderusha, 2018-06-05:
Well of course not that would be super obvious and totally easy and also solve the problem immediately.  Thanks for the gentle guidance :D

closed 2018-06-05 by jneilliii