jneilliii/OctoPrint-BedLevelVisualizer #59: [Feature Request] Display Date of last Mesh-Update

AcHub, 2018-06-09:
I don't refresh my mesh before every print as I noticed only marginal to no changes in bed level heights from print to print in the past. 
But still I want to update the mesh every now and then and it would be nice if somewhere below the graph it would show the Date & Time of the last mesh-update, so I can keep track of it and do an update if it was for example more than 7 days since the last time.

jneilliii, 2018-06-09:
It should show the last time the level was taken if  you hover your mouse over the info icon on the update button.

AcHub, 2018-06-09:
I don't get the hover text displayed with Firefox 60.

Independent of this bug(?), maybe the date/time could be displayed next to the button permanently. There is plenty of room and also you wouldn't have to hassle with hovering over a small icon.