jneilliii/OctoPrint-BedLevelVisualizer #61: Add bed position to the side of the diagram

aplocher, 2018-06-10:
It would be helpful to know which side is the front, back, left and right when viewing the diagram.  Currently when I generate a diagram, I still have no idea which corner it is detecting as being raised up.  In fact, even the default view seems to zoom in on one corner (not the front), so there's no way for me to tell what I'm looking at.

[image]

jneilliii, 2018-06-10:
Except for the dimensions listed along the x and y axis maybe? If you find they are not in the correct orientation to what you expect or figure out from trial and error you can use the flip axis options.

closed 2018-06-10 by jneilliii

aplocher, 2018-06-10:
Yes I see that, but that shows that X would be front OR back, and y would be left OR right, but I still don't know exactly what direction I'm looking.  So in my screenshot, am I looking at the front-left of my bed (Prusa I3 clone [anet a8])?

aplocher, 2018-06-10:
I guess, 0,0 would indicate the first position probed, right?  Which is front-left...

jneilliii, 2018-06-10:
That's very much going to depend on your specific machine and how the firmware is setup, etc.  Typically in a Cartesian system the X axis is left to right and the Y axis is front to back.  So considering your example above the corner that is closest to you is 0,0 which would be the front left corner.

aplocher, 2018-06-10:
Thanks, makes sense.  Since I suppose it's possible people could have their axis setup differently, maybe assuming one side is front/back/left/right isn't always going to be true, but possibly there might be some benefit in allowing custom labels for each side?  Dunno, just an idea.  After talking it through with you (and utilizing the 2 brain cells necessary) it makes sense without the labels now :)

jneilliii, 2018-06-10:
No problem.  I come from an Architecture background so from working within AutoCAD I'm very comfortable with x/y coordinate nomenclature.  It also makes sense in relation to gcode because that's how firmware interprets position. It get's even more complicated when dealing with Delta printers, as 0,0 is in the center of the circular bed.