jneilliii/OctoPrint-BedLevelVisualizer #62: Add support for Marlinkimbra Mk4Duo Bed Topology output

micheleamerica, 2018-06-10:
Would it be possible to include support for the Marlinkimbra (Mk4Duo) firmware?

The output of the G29 T command is different from Marlin and the current version of the plugin is unable to gather the data it needs.

A sample output from Mk4Duo G29 T is as follows:
measured_z = [[ +0.466, +0.550, +0.580, +0.073, +0.448, +0.593, +0.639 ], [ +0.622, -0.054, +0.044, +0.319, +0.599, +0.781, +0.576 ], [ +0.798, +0.024, +0.118, +0.305, +0.546, +0.614, +0.572 ], [ +0.305, -0.027, +0.027, +0.177, +0.336, +0.483, +0.537 ], [ +0.453, -0.226, -0.199, -0.119, +0.087, +0.231, +0.320 ], [ +0.329, -0.301, -0.460, -0.319, -0.110, +0.142, +0.377 ], [ +0.349, +0.428, +0.522, -0.399, +0.384, +0.434, +0.533 ]];

jneilliii, 2018-06-10:
Try other T options, like `G29 T1` or without a T at all `G29`, does that return anything differently?

micheleamerica, 2018-06-10:
There does not seem to make any difference. But here are the outputs without the T parameter, with T1 and with T2. Hope this helps. Thanks.

This is the full output with G29:

<code>
Send: G29Recv: X:0.000000 Y:0.000000 Z:256.897125 E:0.000000 Count Alpha:77791 Beta:77791 Teta:77791Recv: Echo:Bilinear Leveling Grid:Recv: measured_z = [Recv:  [ +0.466, +0.550, +0.580, +0.004, +0.448, +0.593, +0.639 ],Recv:  [ +0.622, -0.076, +0.007, +0.303, +0.583, +0.721, +0.576 ],Recv:  [ +0.798, -0.020, +0.140, +0.302, +0.546, +0.646, +0.572 ],Recv:  [ +0.283, -0.039, +0.014, +0.168, +0.346, +0.486, +0.537 ],Recv:  [ +0.453, -0.223, -0.196, -0.100, +0.087, +0.246, +0.320 ],Recv:  [ +0.329, -0.339, -0.476, -0.285, -0.123, +0.145, +0.377 ],Recv:  [ +0.349, +0.428, +0.522, -0.430, +0.384, +0.434, +0.533 ]Recv: Recv: ];Recv: Echo:Subdivided with CATMULL ROM Leveling Grid:Recv: measured_z = [Recv:  [ +0.46557, +0.48797, +0.51123, +0.53277, +0.55002, +0.57544, +0.60602, +0.61598, +0.57954, +0.45435, +0.26600, +0.09057, +0.00419, +0.05039, +0.18083, +0.33380, +0.44758, +0.50723, +0.54516, +0.57068, +0.59311, +0.61167, +0.62248, +0.63019, +0.63945 ],Recv:  [ +0.50433, +0.46870, +0.43043, +0.39744, +0.37764, +0.38514, +0.41170, +0.43179, +0.41985, +0.34561, +0.23053, +0.12661, +0.08588, +0.14019, +0.25580, +0.38636, +0.48552, +0.54314, +0.58321, +0.61054, +0.62996, +0.63870, +0.63554, +0.62762, +0.62208 ],Recv:  [ +0.54280, +0.44631, +0.34354, +0.25335, +0.19462, +0.18302, +0.20493, +0.23531, +0.24914, +0.22906, +0.19198, +0.16427, +0.17225, +0.23536, +0.33541, +0.44235, +0.52615, +0.58180, +0.62433, +0.65372, +0.66999, +0.66821, +0.65001, +0.62526, +0.60378 ],Recv:  [ +0.58185, +0.43016, +0.26884, +0.12678, +0.03289, +0.00453, +0.02307, +0.06341, +0.10047, +0.12813, +0.15959, +0.19868, +0.24927, +0.31980, +0.40574, +0.49147, +0.56139, +0.61496, +0.65929, +0.69025, +0.70365, +0.69275, +0.66166, +0.62249, +0.58735 ],Recv:  [ +0.62240, +0.42960, +0.22460, +0.04400, -0.07562, -0.11488, -0.09648, -0.04702, +0.00691, +0.06627, +0.14254, +0.22500, +0.30289, +0.37740, +0.45289, +0.52345, +0.58316, +0.63435, +0.67889, +0.71011, +0.72136, +0.70489, +0.66623, +0.61870, +0.57560 ],Recv:  [ +0.68124, +0.46301, +0.23083, +0.02656, -0.10789, -0.14990, -0.12563, -0.06638, -0.00345, +0.06484, +0.15182, +0.24376, +0.32690, +0.40054, +0.47101, +0.53518, +0.58991, +0.63803, +0.67997, +0.70909, +0.71871, +0.70125, +0.66219, +0.61450, +0.57113 ],Recv:  [ +0.75236, +0.52011, +0.27272, +0.05561, -0.08577, -0.12473, -0.09037, -0.02092, +0.04534, +0.10630, +0.17966, +0.25693, +0.32961, +0.39892, +0.46822, +0.53225, +0.58575, +0.63030, +0.66769, +0.69304, +0.70142, +0.68626, +0.65194, +0.60996, +0.57181 ],Recv:  [ +0.80249, +0.56725, +0.31640, +0.09679, -0.04470, -0.07816, -0.03441, +0.04240, +0.10810, +0.15573, +0.20723, +0.26183, +0.31879, +0.38241, +0.45160, +0.51746, +0.57109, +0.61235, +0.64524, +0.66706, +0.67513, +0.66417, +0.63714, +0.60367, +0.57342 ],Recv:  [ +0.79832, +0.57079, +0.32798, +0.11574, -0.02009, -0.04899, -0.00151, +0.07662, +0.13968, +0.17821, +0.21567, +0.25577, +0.30220, +0.36089, +0.42821, +0.49356, +0.54631, +0.58536, +0.61578, +0.63627, +0.64551, +0.63922, +0.61942, +0.59420, +0.57170 ],Recv:  [ +0.70207, +0.50241, +0.28929, +0.10310, -0.01574, -0.04049, +0.00197, +0.07164, +0.12849, +0.16430, +0.19959, +0.23749, +0.28114, +0.33572, +0.39810, +0.45897, +0.50902, +0.54728, +0.57820, +0.60063, +0.61340, +0.61286, +0.60073, +0.58397, +0.56952 ],Recv:  [ +0.54113, +0.38651, +0.22144, +0.07727, -0.01464, -0.03434, -0.00240, +0.05169, +0.09847, +0.13268, +0.16916, +0.20862, +0.25174, +0.30206, +0.35811, +0.41299, +0.45984, +0.49827, +0.53145, +0.55787, +0.57602, +0.58271, +0.57969, +0.57279, +0.56783 ],Recv:  [ +0.37999, +0.26849, +0.14950, +0.04550, -0.02102, -0.03688, -0.01647, +0.02135, +0.05773, +0.09055, +0.12891, +0.17074, +0.21397, +0.26046, +0.31050, +0.35972, +0.40373, +0.44282, +0.47886, +0.50977, +0.53351, +0.54730, +0.55297, +0.55540, +0.55945 ],Recv:  [ +0.28314, +0.19378, +0.09855, +0.01506, -0.03907, -0.05447, -0.04207, -0.01481, +0.01439, +0.04515, +0.08335, +0.12542, +0.16784, +0.21145, +0.25755, +0.30324, +0.34563, +0.38545, +0.42375, +0.45810, +0.48606, +0.50513, +0.51722, +0.52653, +0.53723 ],Recv:  [ +0.28424, +0.18498, +0.07942, -0.01353, -0.07495, -0.09444, -0.08387, -0.05774, -0.03054, -0.00334, +0.03065, +0.06875, +0.10825, +0.15029, +0.19568, +0.24126, +0.28385, +0.32417, +0.36338, +0.39901, +0.42855, +0.44963, +0.46415, +0.47604, +0.48925 ],Recv:  [ +0.34340, +0.21438, +0.07735, -0.04368, -0.12473, -0.15152, -0.13949, -0.10915, -0.08101, -0.05820, -0.03056, +0.00133, +0.03690, +0.07831, +0.12509, +0.17274, +0.21677, +0.25763, +0.29718, +0.33299, +0.36262, +0.38366, +0.39800, +0.40965, +0.42264 ],Recv:  [ +0.41482, +0.25188, +0.07904, -0.07399, -0.17753, -0.21316, -0.20021, -0.16562, -0.13631, -0.11739, -0.09518, -0.06846, -0.03602, +0.00521, +0.05365, +0.10361, +0.14941, +0.19132, +0.23178, +0.26833, +0.29851, +0.31985, +0.33426, +0.34589, +0.35892 ],Recv:  [ +0.45266, +0.26742, +0.07121, -0.10307, -0.22251, -0.26678, -0.25729, -0.22372, -0.19575, -0.17887, -0.15805, -0.13221, -0.10029, -0.05931, -0.01075, +0.03983, +0.08682, +0.13069, +0.17377, +0.21329, +0.24647, +0.27083, +0.28842, +0.30331, +0.31955 ],Recv:  [ +0.43907, +0.24362, +0.03718, -0.14730, -0.27694, -0.33333, -0.33723, -0.31446, -0.29082, -0.26776, -0.23333, -0.19323, -0.15317, -0.11390, -0.07262, -0.02977, +0.01423, +0.06162, +0.111 [... cut ...] 8156, -0.39160, -0.18131, +0.05495, +0.22280, +0.29167, +0.31369, +0.31855, +0.33596, +0.37185, +0.41105, +0.45157, +0.49143 ],Recv:  [ +0.34913, +0.36854, +0.38771, +0.40735, +0.42815, +0.47700, +0.54463, +0.57749, +0.52204, +0.29773, -0.04479, -0.36227, -0.51147, -0.40348, -0.13162, +0.17410, +0.38367, +0.45446, +0.45846, +0.43759, +0.43373, +0.45502, +0.48013, +0.50676, +0.53263 ]Recv: Recv: ];Recv: X:0.000000 Y:105.000000 Z:30.369501 E:0.000000 Count Alpha:29719 Beta:29809 Teta:48906
</code>

jneilliii, 2018-06-10:
This will be difficult for me to accomplish because of the reporting output of your firmware.  The main problem is that every version of the report gives the same results, which is two sets of data, the simple and subdivided data. This is divergent to other firmware as those would give different reports based on the T parameter. 

I will consider making this work in the future, but you may want to also post a feature request to the firmware's repository to report `G29 T1` as a pure csv output instead of the JSON format they appear to be sending back now. This would be consistent with how the latest Marlin reports with UBL.  I realize with the latest 1.1.8 firmware the CSV report is not actually comma separated, but it's a pure data model rather than a JSON formatted one.

```
Send: G29 T1
Recv: echo:Home XYZ first
Recv: 
Recv: Bed Topography Report for CSV:
Recv: 
Recv: -0.318	-0.150	-0.087	-0.115	-0.025	0.211	0.272	0.413	0.554	NAN
Recv: -0.169	-0.220	-0.238	-0.184	0.033	0.056	0.286	0.313	0.341	NAN
Recv: -0.324	-0.119	-0.123	-0.290	-0.012	0.046	0.251	0.360	0.468	NAN
Recv: -0.122	-0.033	-0.039	-0.071	-0.057	-0.064	0.089	0.413	0.737	NAN
Recv: -0.006	-0.083	-0.071	-0.060	-0.048	-0.064	0.060	0.261	0.461	NAN
Recv: 0.082	0.026	-0.017	-0.231	0.063	0.056	0.055	0.324	0.592	NAN
Recv: 0.191	0.155	0.018	0.035	0.229	0.131	0.433	0.423	0.423	NAN
Recv: 0.025	0.109	0.222	0.130	0.223	0.291	0.211	0.567	0.924	NAN
Recv: 0.135	0.336	0.190	0.050	0.219	0.289	0.393	0.555	0.716	NAN
Recv: 0.407	0.296	0.226	0.277	0.365	0.446	0.404	0.524	0.644	NAN
Recv: ok P15 B3
```