jneilliii/OctoPrint-BedLevelVisualizer #63: Today's update 6/10/18

bill-orange, 2018-06-10:
I am running a PI B+ I loaded today's update to bed visualizer.  I am running a PI B+.  It displayed the existing mesh okay but when I tried updating the mesh, Octopi hung.  The extruder did not move and the Updating mesch message was displayed forever.  The only way I found to recover was to reboot the PI.  I tried it twice with the same result just to be sure.  I have not truied uninstalling the App and reinstalling it.  I needed the machine this afternoon.

bill-orange, 2018-06-10:
I can try uninstalling and reinstalling when the machine is not busy.

jneilliii, 2018-06-10:
Check the logs too to see if there are any other errors. Copy/paste the terminal tab wrapped with 3 back ticks on each side so the formatting stays in tact.

bill-orange, 2018-06-10:
Will do. Current print job has a ways to go.

bill-orange, 2018-06-10:
Where are the logs stored?

jneilliii, 2018-06-10:
In the octoprint's settings there is a logging section where you can download octoprint.log.

bill-orange, 2018-06-11:
Hopefully will check later today.  may be the same problem as @roofg

jneilliii, 2018-06-11:
If it's the same issue as @roofg then setting the gcode commands in the plugin settings to something like the following should fix it. The key point on the GCODE command is that the flag `@BEDLEVELVISUALIER` needs to be as close to the reporting command as possible to start the data collection process correctly. It stops collecting when it receives an ok response from the printer after that command.  So if you don't include that flag in your settings it will stop collecting after the first command in the settings was sent.  If that's not a bed leveling report then it never gets the data to render correctly.

```
G28
M155 S30
@BEDLEVELVISUALIZER
G29 T
M155 S3
```

bill-orange, 2018-06-12:
I will try to test tomorrow.  Here's my current script now. Do I need the M420 V
G28
M155 S40
@BEDLEVELVISUALIZER
G29 T
M155 S3
M420 V

jneilliii, 2018-06-12:
No, probably not.  also, make sure the `@BEDLEVELVISUALIZER` is all caps.

jneilliii, 2018-07-29:
In order to debug your issue, I'll need more details. For example, what are your gcode commands setting used and what is returned in the terminal tab when those commands are run?

closed 2018-08-19 by stale[bot]