jneilliii/OctoPrint-BedLevelVisualizer #65: Does BedLevelVisualizer work with Repetier firmware?

bumbalu, 2018-06-12:
And if it does what would be the GCODE? G29 T produces different output in Repetier than in Marlin. 
Thank you.

jneilliii, 2018-06-12:
I would have to see the output of the G29 T command from our printer to say for sure.

bumbalu, 2018-06-12:
This is what is produced by G29 T command, it measures only 3 points (corners) of the table height:
Send: N18 G29 T*95
Recv: ok 18
Recv: Info:Autoleveling disabled
Recv: Z-probe:20.580 X:-0.01 Y:0.00
Recv: Z-probe:20.588 X:159.00 Y:0.00
Recv: Z-probe:20.560 X:-0.01 Y:144.00
Recv: Z-probe average height:20.58
Recv: Info:Autoleveling enabled

jneilliii, 2018-06-12:
Unfortunately, that format will not work with this plugin.

closed 2018-06-12 by jneilliii

bumbalu, 2018-06-12:
Ok, thank you.

jneilliii, 2018-06-12:
What does your printer report with `G33 L0`?