jneilliii/OctoPrint-BedLevelVisualizer #66: Settings for a Prusa MK 2.5 / MK3

Petclaud, 2018-06-12:
Hi, this looks like a real interesting plugin to have.
Unfortunately I do not how to make this work on an Prusa MK2.5 or MK3.

If I look here: https://github.com/prusa3d/Prusa-Firmware/wiki/Supported-G-codes I see that G80 is the mesh bed leveling command. If I fill this command in the Bed Level Visualizer and run it, the printer does perform a 9 point mesh bed leveling, however after that the plugin window keeps on telling that it is waiting for information. If I look in the Terminal TAB during mesh bed leveling I see this:

[image]

Sorry I am not a GCODE expert. Do you have any idea if an how I can get this to work on the original Prusa's...?

jneilliii, 2018-06-12:
Try using the following.

```
G80
@BEDLEVELVISUALIZER
G81
```

closed 2018-06-12 by jneilliii

Petclaud, 2018-06-24:
First of all, I am very sorry for this way to late reply.... Your response was buried under many other GitHub notifications and I did not return to this thread till now. 

Thank you very much for the solution!!! This is is a briljant plugin! Keep up the good work.

tobymurray, 2018-10-08:
What does `@BEDLEVELVISUALIZER` do in this scenario? How is it different from just G80 G81?

jneilliii, 2018-10-08:
That command is an instruction for the plugin to start capturing the responses from the printer and has to be used in conjunction with your leveling report command. I don't know id the prusa firmware requires the G80 command to be run prior to issuing a G81 command but if it's not and the offsets are stored in eeprom you can just use the following. 

```
@BELEVELVISUALIZER
G81
```