jneilliii/OctoPrint-BedLevelVisualizer #70: problem with i3 clone and marlin 1.1.8

sgoudelis, 2018-06-21:
Hello,

The G29 T command gives this:

```
Send: G29 T
Recv: State: On
Recv: Num X,Y: 3,3
Recv: Z offset: 0.00000
Recv: Measured points:
Recv:         0        1        2
Recv:  0 +5.00000 +5.00000 +5.00000
Recv:  1 +5.00000 +5.00000 +5.00000
Recv:  2 +5.00000 +5.00000 +5.00000
Recv: 
Recv: X:80.00 Y:111.00 Z:0.00 E:0.00 Count X:8000 Y:11100 Z:2000
Recv: ok
```

Not sure if this is ok. I am assuming that those 5s are not a valid value right ?
Do you have any idea what might be going on ?

sgoudelis, 2018-06-21:
I did work when I switched to ABL and not Mesh bed leveling.

jneilliii, 2018-06-21:
Yeah, I'm not so sure what those 5s are. Sounds like you got it working though with the other type of leveling.

closed 2018-06-21 by jneilliii