jneilliii/OctoPrint-BedLevelVisualizer #71: Crashes when no levelling data present

paulr4, 2018-07-06:
The plugin gets stuck endlessly when there's no levelling data to display, i.e. when Marlin returns the following:

Recv: Mesh bed leveling has no data.

Perhaps a timeout would be a good idea.

Great plugin otherwise!

wwolkers, 2018-07-16:
I have the same issue when the bed leveling stops for some reason, no data makes the plugin hang

gravesfamily, 2018-11-11:
Or, if I've just started up my device (I have one switch that turns on both the printer and the octopi) and I have forgotten to level the printer first (I was just going to look at the existing previous level data, which does exist) then it goes into "please wait, retrieving mesh" and won't unstick itself even after I do a G28.

Love this visualizer, by the way. It's made it much easier to figure out which corner(s) need to be moved and which way.

jneilliii, 2018-11-17:
I've just pushed a new branch for testing that incorporates the timeout concept to unblock the UI that gets blocked during processing. Defaults to 1 minute after pressing the button.  If processing isn't able to complete within that time frame an info message will be displayed indicating such.  Processing will still continue if there isn't a configuration error and the surface graphing will regenerate and the timestamp will update once the processing finishes.

You can test by installing the plugin using the URL below in plugin manager.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.7.zip

@gravesfamily, if the plugin isn't working for you please post a new issue making sure to include your octoprint.log file, any errors in your browser's developer console, printer firmware (including version), and your gcode settings.

PR #96 (2018-11-25): 0.1.7
**Added**
  - New timeout setting that will unlock `processing` UI after given seconds.  If processing takes longer to complete due to full probing requirements or some other reason set this to a higher value. Resolves #71.
  - New button in graph toolbar that will enable click to move nozzle operations. Click a point in the graph and the gcode command will be sent to the printer to move to that point.
closed 2018-11-25 by jneilliii