jneilliii/OctoPrint-BedLevelVisualizer #72: Can't install BedLevelVisualizer on Windows 10 PC

3-DPO, 2018-07-15:
I'm new to 3D printing.  I don't find any mention of this specific issue (I spent hours trying to find one!)  I installed OctoPrint so I could get your BedLevelVisualizer Plugin (heard great things!)  But the Plug-in install failed with an "unknown" error (image below).  I am running the following combination of firmware/software.  

If I can give you more information, please let me know.  (BTW, I saw this issue https://github.com/foosel/OctoPrint/issues/2735 which mentions a similar problem with your Telegram and TPLink  Plug-ins.  That issue may get directed to you eventually)  

Thank you for your help.

**My setup:**
PC operating system: Windows 10 Home Version 1803 Build 17134.165
Printer: Prusa i3 MK3
Printer Firmware: 3.3.0-830
OctoPrint: 1.3.8 (with the following installs):

- Python version 2.7.14

- Microsoft Visual C++ Compiler for Python 2.7

- virtualenv using pip install virtualenv: pip install virtualenv (this installed at version 9.0.1, but as suggested in the log entry, I upgraded to version 10.0.1)

[image]

jneilliii, 2018-07-15:
That unknown message typically is registered to a python error on install. Look inside octoprint.log for a traceback error for me please, or upload your log here for me to look.

3-DPO, 2018-07-15:
I find a several traceback errors (tried the install more than once).  Not knowing exactly what you want to look at, I'm attaching a copy of the octoprint.log.  Thank you.
[octoprint.log.copy.txt](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/2196151/octoprint.log.copy.txt)

jneilliii, 2018-07-16:
Looks like you are effected by an issue with sarge and your pip versions I think. I've seen some posts on octoprint issue tracker related to this if I remember correctly, or possibly on the community forum. Pip version 10 is probably at the root of the issue here.

3-DPO, 2018-07-16:
I see!  I found this https://discourse.octoprint.org/t/installing-a-plugin-shows-me-unknown-in-the-popup-i-cant-uninstall-plugins-through-the-plugin-manager-octoprint-up-to-1-3-8-octopi-0-15-0-or-pip-10-x/1746 which says "A workaround ... is to downgrade pip to version 9.0.3, restart your server and reinstall the affected plugin through the plugin manager."  OctoPrint 1.3.9 will contain a fix for this apparently.  I'll try downgrading pip tomorrow and let you know how I make out.

jneilliii, 2018-07-16:
Also, look here...

https://github.com/foosel/OctoPrint/issues/2687

3-DPO, 2018-07-16:
Thanks for the above tip.  mexusbg solved their issue by downgrading sarge.  I downgraded pip to 9.0.3, and sarge to 0.1.4.  Cleared my browser caches, launched OctoPrint, and tried to reinstall this Plug-in but got the exact same error as before. Dang!  

I also followed up on foosel/octoprint#2735 and see that there is a new FAQ at https://discourse.octoprint.org/t/i-cant-install-uninstall-a-plugin-i-get-something-about-an-unexpected-keyword-argument-async/2801/1  These instructions say to downgrade pip and sarge from the OctoPrint directory.  I had done these steps from the Python27 directory, so I ran them from OctoPrint - it said I already **was** at pip 9.03 and sarge 0.1.4  

So now I'm at a loss.  Could it be that I'm not running on a Pi but on a laptop?  I don't see why/how that would matter. Ideas?

jneilliii, 2018-07-16:
Sorry, no idea on the differences here. I'm not a python expert.

jneilliii, 2018-07-16:
Ok, I just read that FAQ post.  You should be able to follow the same steps listed there.  So you would start a command prompt and type the following:
```
cd <path to octoprint folder>\venv
``` 
In my case c:\OctoPrint\venv\
```
activate.bat
pip install sarge==0.1.4
```
I see you stated that you tried those steps, so not 100% sure here.  I have a windows install here, I'm going to give it a try and see if I can reproduce your errors and will report back my findings.

jneilliii, 2018-07-16:
If you type the following after running the above commands (except the pip install line) what comes back for you?

```
pip freeze
```

jneilliii, 2018-07-16:
I just went through the install process on my windows install without issues, so I do think this is something related to your environment.  I would seek help in the Community Forum, there are some real sharp guys over there that will be able to assist better than myself probably.

closed 2018-07-16 by jneilliii

3-DPO, 2018-07-18:
Thank you jneilliii!  (Sorry for the delay getting back to you - my life got in the way!)

It appears that I didn't actually execute "activate.bat"  - when I entered it on a separate command line as you specified above, it came back with a "(venv) C:\OctoPrint\venv\Scripts" prompt (something I didn't get before).    _Then_ I ran the downgrade of sarge (pip install sarge==0.1.4).  I had run this last step before, but never from this prompt.

I figured I also should rerun the steps to downgrade pip to 9.0.3  (python.exe -m pip install pip==9.0.3) That worked too!

Next I reran the Plug-in Install and it worked!  The details above are nothing new for you, I'm sure, but maybe they will help someone struggling with this.  I haven't been a code monkey for a _long time_ - I'm a little rusty!  Thanks again for your help!