jneilliii/OctoPrint-BedLevelVisualizer #74: Plugin doesn't work for Workbench Apprentice

chancrescolex, 2018-07-27:
The data returned by my printers z-probe doesn't look exactly like what's in your examples so maybe that's the issue.

Here's the gcode I'm sending:

`G21; set units to mm
G90; set to absolute positioning
G28 X0 Y0; home x and y axes
G1 Z20 F200; lower bed for clearance
G1 X0 Y14 F5000; move to front left corner
G92 X0 Y0; set correct bed origin
M155 S30
@BEDLEVELVISUALIZER
G29; detailed z-probe
M155 S3
G1 Z10 F200; lower bed for clearance`

And here is the printers response:

`Send: M155 S30
Recv: ok
Send: G29
Recv: Bed x: 40.00 y: 20.00 z: 7.31
Recv: Bed x: 80.00 y: 20.00 z: 7.38
Recv: Bed x: 120.00 y: 20.00 z: 7.35
Recv: Bed x: 120.00 y: 70.00 z: 7.37
Recv: Bed x: 80.00 y: 70.00 z: 7.41
Recv: Bed x: 40.00 y: 70.00 z: 7.35
Recv: Bed x: 40.00 y: 120.00 z: 7.33
Recv: Bed x: 80.00 y: 120.00 z: 7.36
Recv: Bed x: 120.00 y: 120.00 z: 7.34
Recv: Eqn coefficients: a: 0.00 b: -0.00 d: 7.33
Recv: planeNormal x: -0.00 y: 0.00 z: 1.00
Recv: ok
Send: M155 S3
Recv: echo:endstops hit:  Z:7.34
Recv: ok
Send: G1 Z10 F200
Recv: ok`

The plugin just hangs and says "Please wait, retrieving current mesh."

chancrescolex, 2018-07-27:
Also tried `G29 T` instead of just `G29` but it sends back the same data. I think it's due to my printers custom firmware.

jneilliii, 2018-07-27:
Yeah, it looks like either an older Marlin format or possibly a smoothieware firmware leveling report.  The plugin currently can't utilize that format to extract the data necessary to render.  Several people have had that same issue, I just haven't had the time to update the plugin to support it yet.

jneilliii, 2018-07-27:
After going back and reviewing some other issues I remembered that I was able to get this to work with one of the Smoothieware leveling reports over on #22. This report is very similar to that so it probably wouldn't take much for me to get this to work for you. I'll see what I can do this weekend.  The only difference is your replies with `Bed...` instead of `DEBUG...` like in the other issue.

chancrescolex, 2018-07-27:
What are your thoughts on supporting custom report formats by having a config box where the user enters their report format with variables so the plugin knows exactly where to find the data it needs? For example, mine would be like the following:

```Recv: Bed x: @X1 y: @Y1 z: @Z1
Recv: Bed x: @X2 y: @Y2 z: @Z2
Recv: Bed x: @X3 y: @Y3 z: @Y3
(And so on)
```

jneilliii, 2018-07-29:
Be aware that the points won't match exactly the x/y coordinates as it interprets that from the overall size of the bed rather than the inset point. It should be close enough to get an idea of tilt/warping though.

chancrescolex, 2018-08-06:
Sorry I didn’t get a chance to test this yet. I’ll try to test it early this week and report back.