jneilliii/OctoPrint-BedLevelVisualizer #75: Todays Update 08062018 hangs ...

LachmannM, 2018-08-06:
I tried updating the plugin from within octopis plugin manager but installation hangs with 
Running setup.py install for numpy: still running...

The same happened after removing the plugin and adding it again ... this time during the installation.

I've added the log file ...

Best regards,
  Michael
[plugin_pluginmanager_console.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/2262502/plugin_pluginmanager_console.log)

akurz42, 2018-08-06:
Exactly the same issue here.

c3r63ru5, 2018-08-06:
I checked and its cc1 running in the background that took approx 15mins on my raspi3. Just give it some time...

jneilliii, 2018-08-06:
Agreed, it took quite some time for this to update for me as well as it has to pull in an additional library and compile it.  The additional library was for numpy. It will eventually complete the update.

closed 2018-08-06 by jneilliii