jneilliii/OctoPrint-BedLevelVisualizer #76: v0.1.3 no longer works for me

OutsourcedGuru, 2018-08-06:
Updated to 0.1.3 today. The compilation of `numpy` took a full 24 minutes, btw. Clicking the Update button yields a forever-spinning wait status: "Please wait, retrieving current mesh.".  It looks like the Terminal is now unresponsive; it won't accept an M115 command.

Robo C2

```
Send: G29 T1
Recv: G29 Auto Bed Leveling
Recv: echo:busy: processing
Printer seems to support the busy protocol, adjusting timeouts and setting busy interval accordingly
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Eqn coefficients: a: 0.00158019 b: -0.00174903 d: 4.77895975
Recv: 
Recv: Bed Height Topography:
Recv:    +--- BACK --+
Recv:    |           |
Recv:  L |    (+)    | R
Recv:  E |           | I
Recv:  F | (-) N (+) | G
Recv:  T |           | H
Recv:    |    (-)    | T
Recv:    |           |
Recv:    O-- FRONT --+
Recv:  (0,0)
Recv:  -0.23208 -0.04833 +0.20042
Recv:  -0.07583 -0.04458 -0.17083
Recv:  +0.07167 +0.06292 +0.23667
Recv: 
Recv: 
Recv: 
Recv: Bed Level Correction Matrix:
Recv: +0.999999 +0.000000 +0.001580
Recv: +0.000003 +0.999998 -0.001749
Recv: -0.001580 +0.001749 +0.999997
Recv: X:116.01 Y:85.99 Z:4.91 E:0.00 Count X: 9280 Y:6880 Z:3958
Recv: ok
Send: M113 S2
Recv: ok
```

jneilliii, 2018-08-06:
Anything weird in octoprint.log?

jneilliii, 2018-08-06:
I think I know what was causing the issue.  Your report was getting flagged as the new format I was accounting for and therefore wasn't processing correctly.  Try installing 0.1.4 from the url below in plugin manager and let me know if it's still an issue.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.4.zip

OutsourcedGuru, 2018-08-08:
Will do... (after this print job).

jneilliii, 2018-08-13:
Thanks.  officially released version 0.1.4.