jneilliii/OctoPrint-BedLevelVisualizer #78: Unable to install to clean copy of Octoprint

sponge14, 2018-08-21:
At first, I was unable to update to the new version of the bed level.  So I completely reinstalled Octoprint, and the very first thing I tried to install was the bed level plugin.  I am unable to install, I get an error, it says to check the following log:
[octoprint (1).log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/2304345/octoprint.1.log)

sponge14, 2018-08-21:
[plugin_pluginmanager_console (1).log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/2304612/plugin_pluginmanager_console.1.log)

I have tried to load numpy twice with the pip install, both times Putty says that it disconnected so I am not sure if it finished.  I tried to install the plugin again, and this is the log file from latest failed attempt.  Thanks!

jneilliii, 2018-08-21:
Yeah, looks like numpy download/build is not completing at all.  Not sure if there is anything I can do for that.  It took a considerable amount of time for me on my Pi3, somewhere around 20-30 minutes if I remember correctly.  You could always just install the old version and not upgrade, just use the url below in Plugin Manager.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.2.zip

sponge14, 2018-08-21:
Using the older version worked perfectly, thank you!

closed 2018-08-24 by jneilliii