jneilliii/OctoPrint-BedLevelVisualizer #80: Add an option to treat center as zero

repvik, 2018-09-02:
I've mounted my bed according to this guide: https://github.com/PrusaOwners/prusaowners/wiki/Bed_Leveling_without_Wave_Springs

The center support is static, perimeter supports are adjustable. It'd be very nice if the plugin could add an option to treat dead center as 0 and recalculate other points as relative to that point.

jneilliii, 2018-09-02:
That is actually how I have the plugin working for circular beds. Adding an option to use the same logic would not be too difficult I don't think. I'll see what I can do.

jneilliii, 2018-09-02:
It was extremely easy to integrate this option.  Please install and test for me and if everything works for you as expected I'll release an official version.  Install using the following url in plugin manager. Once installed go into settings and check the box to "Use Center Origin" and then press Update on the Bed Visualizer tab.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.5.zip

[image]

repvik, 2018-09-02:
Not entirely what I expected. It looks like the X-axis is 0 at center, not the Z-axis. 

[image]

repvik, 2018-09-02:
Here's the graph data:

> -125	0.13922	0.14779	0.15284	0.15438	0.15240	0.14690	0.13788	-120
-83	0.14360	0.14951	0.15314	0.15449	0.15355	0.15033	0.14483	-83
-42	0.14674	0.15044	0.15292	0.15420	0.15425	0.15310	0.15073	-47
0	0.14866	0.15057	0.15218	0.15349	0.15450	0.15520	0.15560	-10
42	0.14934	0.14991	0.15092	0.15238	0.15429	0.15664	0.15944	27
83	0.14878	0.14844	0.14914	0.15086	0.15362	0.15741	0.16223	63
125	0.14699	0.14618	0.14683	0.14894	0.15250	0.15752	0.16399	100

jneilliii, 2018-09-02:
Can you copy paste the results in the terminal tab when you run the gcode command to report the data. I think it's G31 maybe. Make sure to surround it in code comments using three back ticks before and after please.

repvik, 2018-09-02:
Just started a 3hr print, will report back later.

jneilliii, 2018-09-02:
Sure, no problem.  I'll be offline for the rest of the night, so no rush.

repvik, 2018-09-02:
```
Send: M140 S0
Recv: Measured points:
Recv:   0.13988  0.13547  0.13399  0.13544  0.13981  0.14710  0.15733
Recv:   0.14329  0.13725  0.13429  0.13441  0.13762  0.14391  0.15328
Recv:   0.14551  0.13929  0.13582  0.13508  0.13708  0.14182  0.14930
Recv:   0.14655  0.14161  0.13858  0.13744  0.13819  0.14084  0.14538
Recv:   0.14640  0.14421  0.14257  0.14149  0.14095  0.14096  0.14152
Recv:   0.14507  0.14708  0.14780  0.14723  0.14536  0.14219  0.13773
Recv:   0.14255  0.15023  0.15427  0.15466  0.15141  0.14452  0.13399
```

The chart data:
```
-125	0.14255	0.15023	0.15427	0.15466	0.15141	0.14452	0.13399	-120
-83	0.14507	0.14708	0.14780	0.14723	0.14536	0.14219	0.13773	-83
-42	0.14640	0.14421	0.14257	0.14149	0.14095	0.14096	0.14152	-47
0	0.14655	0.14161	0.13858	0.13744	0.13819	0.14084	0.14538	-10
42	0.14551	0.13929	0.13582	0.13508	0.13708	0.14182	0.14930	27
83	0.14329	0.13725	0.13429	0.13441	0.13762	0.14391	0.15328	63
125	0.13988	0.13547	0.13399	0.13544	0.13981	0.14710	0.15733	100
```

I'm inverting the Y axis, since Prusa presents the values so the bed is flipped.

jneilliii, 2018-09-03:
So, yeah based on your measured points there is no Z=0. I assume that is because you didn't apply the firmware hack mentioned in that post you linked? Sorry I didn't catch the fact that you wanted to adjust the leveling data based on the value of the center offset.  That will take a considerable amount of additional math work to make that happen.  I think because I've added the numpy library to this plugin it won't be too difficult with some of it's functions, I'm just not familiar enough with the library so it will take some research time on my part to make that happen.

jneilliii, 2018-09-03:
Just pushed an update to that same test version, reinstall using the previous url. Once installed enable the option for Relative offset distances and you should see what you were looking for.

[image]

repvik, 2018-09-03:
I'm sorry if I was unclear. The idea was to recalculate so center was 0. Saves everyone from recompiling the firmware each time they upgrade. 

I've tested it, and assuming the plugin actually updated, it's not quite there yet:
```
Recv:   0.15011  0.15923  0.16479  0.16677  0.16519  0.16004  0.15133
Recv:   0.15580  0.16447  0.16978  0.17173  0.17031  0.16552  0.15737
Recv:   0.15948  0.16677  0.17120  0.17280  0.17156  0.16747  0.16054
Recv:   0.16116  0.16611  0.16905  0.17000  0.16894  0.16589  0.16083
Recv:   0.16084  0.16250  0.16333  0.16332  0.16247  0.16079  0.15826
Recv:   0.15850  0.15595  0.15404  0.15277  0.15214  0.15215  0.15281
Recv:   0.15417  0.14644  0.14117  0.13833  0.13794  0.14000  0.14450
```

```
0.15417	0.14644	0.14117	0.13833	0.13794	0.14000	0.14450
0.15850	0.15595	0.15404	0.15277	0.15214	0.15215	0.15281
0.16084	0.16250	0.16333	0.16332	0.16247	0.16079	0.15826
0.16116	0.16611	0.16905	0.17000	0.16894	0.16589	0.16083
0.15948	0.16677	0.17120	0.17280	0.17156	0.16747	0.16054
0.15580	0.16447	0.16978	0.17173	0.17031	0.16552	0.15737
0.15011	0.15923	0.16479	0.16677	0.16519	0.16004	0.15133
```

As I understand it, this requires finding the midpoint, and subtracting this from all given values if the option is checked.

jneilliii, 2018-09-03:
That's exactly what it should be doing now.  I just mimicked your results from terminal on the virtual printer on my octoprint install and ran it through the processing and this is what I get for the mesh data.

```
-0.01989000000000002	-0.010770000000000002	-0.00521000000000002	-0.0032300000000000106	-0.004810000000000009	-0.009960000000000024	-0.01867000000000002
-0.014200000000000018	-0.005530000000000007	-0.00022000000000002573	0.0017299999999999816	0.00030999999999997696	-0.004480000000000012	-0.012630000000000002
-0.010520000000000002	-0.0032300000000000106	0.0011999999999999789	0.002799999999999997	0.001559999999999978	-0.0025300000000000045	-0.009460000000000024
-0.008840000000000015	-0.0038900000000000046	-0.0009500000000000064	0	-0.0010600000000000054	-0.0041100000000000025	-0.009170000000000011
-0.009160000000000001	-0.007500000000000007	-0.006670000000000009	-0.006680000000000019	-0.007530000000000009	-0.009210000000000024	-0.01174
-0.01150000000000001	-0.014050000000000007	-0.015960000000000002	-0.017230000000000023	-0.017860000000000015	-0.017850000000000005	-0.01719000000000001
-0.01583000000000001	-0.023560000000000025	-0.028830000000000022	-0.031670000000000004	-0.032060000000000005	-0.03	-0.025500000000000023
```

Sometimes when updating the same version you have to force a browser refresh.  Try restarting OctoPrint and then when the browser reconnects if it doesn't prompt to reload hold down the ctrl key and press the f5 button on your keyboard.

repvik, 2018-09-03:
Thank you, works now.

jneilliii, 2018-09-04:
Closed with official release of 0.1.5.

closed 2018-09-04 by jneilliii