jneilliii/OctoPrint-BedLevelVisualizer #81: Cant install

ChrisWLeon, 2018-09-04:
I attempted to install Bed visualiser. Took a long time "hours" so i cancelled it. I later tried to reinstall and it progressed all the way through in a minutes of so to asking to restart octoprint. Then it would display the message to reconnect, but instead of progressing to the reload page it then would cancel. Restarting octotprint manually and Bed Visualiser is not installed?

jneilliii, 2018-09-05:
Sounds like an incomplete install. You may have to go through the process to manually remove using pip. And try again.

saikek, 2018-09-09:
Same here. Removed previous install & installed from scratch.
It hangs on bdist_wheel step:

```bash
09:28 orangepizero saikek /opt/octoprint/venv/bin ./pip install "https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip"
Collecting https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip
  Downloading https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip
     - 1.0MB 2.8MB/s
Requirement already satisfied: OctoPrint in /opt/octoprint/venv/lib/python2.7/site-packages (from Bed-Visualizer==0.1.6) (1.3.9)
Collecting numpy (from Bed-Visualizer==0.1.6)
  Downloading https://files.pythonhosted.org/packages/65/ab/4dfcc20234fae12ee40c714b98077d6e3a10652496bd1488fa4828529b22/numpy-1.15.1.zip (4.5MB)
    100% |████████████████████████████████| 4.5MB 472kB/s
Requirement already satisfied: pyserial<3.5,>=3.4 in /opt/octoprint/venv/lib/python2.7/site-packages/pyserial-3.4-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (3.4)
Requirement already satisfied: pylru<1.1,>=1.0.9 in /opt/octoprint/venv/lib/python2.7/site-packages/pylru-1.0.9-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (1.0.9)
Requirement already satisfied: semantic-version<2.5,>=2.4.2 in /opt/octoprint/venv/lib/python2.7/site-packages/semantic_version-2.4.2-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (2.4.2)
Requirement already satisfied: future<0.16,>=0.15.2 in /opt/octoprint/venv/lib/python2.7/site-packages/future-0.15.2-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (0.15.2)
Requirement already satisfied: scandir<1.4,>=1.3 in /opt/octoprint/venv/lib/python2.7/site-packages/scandir-1.3-py2.7-linux-armv7l.egg (from OctoPrint->Bed-Visualizer==0.1.6) (1.3)
Requirement already satisfied: sarge==0.1.4 in /opt/octoprint/venv/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (0.1.4)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /opt/octoprint/venv/lib/python2.7/site-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (1.6.5)
Requirement already satisfied: futures<3.2,>=3.1.1 in /opt/octoprint/venv/lib/python2.7/site-packages/futures-3.1.1-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (3.1.1)
Requirement already satisfied: rsa<3.3,>=3.2.3 in /opt/octoprint/venv/lib/python2.7/site-packages/rsa-3.2.3-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (3.2.3)
Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /opt/octoprint/venv/lib/python2.7/site-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (0.10)
Requirement already satisfied: netaddr<0.8,>=0.7.19 in /opt/octoprint/venv/lib/python2.7/site-packages/netaddr-0.7.19-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (0.7.19)
Requirement already satisfied: tornado==4.5.3 in /opt/octoprint/venv/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6) (4.5.3)
Requirement already satisfied: flask<0.11,>=0.10.1 in /opt/octoprint/venv/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (0.10.1)
Requirement already satisfied: Click<6.3,>=6.2 in /opt/octoprint/venv/lib/python2.7/site-packages/click-6.2-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (6.2)
Requirement already satisfied: chainmap<1.1,>=1.0.2 in /opt/octoprint/venv/lib/python2.7/site-packages/chainmap-1.0.2-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (1.0.2)
Requirement already satisfied: wrapt<1.11,>=1.10.11 in /opt/octoprint/venv/lib/python2.7/site-packages/wrapt-1.10.11-py2.7-linux-armv7l.egg (from OctoPrint->Bed-Visualizer==0.1.6) (1.10.11)
Requirement already satisfied: PyYAML<3.11,>=3.10 in /opt/octoprint/venv/lib/python2.7/site-packages/PyYAML-3.10-py2.7-linux-armv7l.egg (from OctoPrint->Bed-Visualizer==0.1.6) (3.10)
Requirement already satisfied: monotonic<1.4,>=1.3 in /opt/octoprint/venv/lib/python2.7/site-packages/monotonic-1.3-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (1.3)
Requirement already satisfied: frozendict<1.3,>=1.2 in /opt/octoprint/venv/lib/python2.7/site-packages/frozendict-1.2-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (1.2)
Requirement already satisfied: netifaces<0.11,>=0.10.6 in /opt/octoprint/venv/lib/python2.7/site-packages/netifaces-0.10.7-py2.7-linux-armv7l.egg (from OctoPrint->Bed-Visualizer==0.1.6) (0.10.7)
Requirement already satisfied: pkginfo<1.5,>=1.4.2 in /opt/octoprint/venv/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6) (1.4.2)
Requirement already satisfied: watchdog<0.9,>=0.8.3 in /opt/octoprint/venv/lib/python2.7/site-packages/watchdog-0.8.3-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (0.8.3)
Requirement already satisfied: python-dateutil<2.7,>=2.6.1 in /opt/octoprint/venv/lib/python2.7/site-packages/python_dateutil-2.6.1-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (2.6.1)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in /opt/octoprint/venv/lib/python2.7/site-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (5.2.1)
Requirement already satisfied: requests<3,>=2.18.4 in /opt/octoprint/venv/lib/python2.7/site-packages/requests-2.19.1-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (2.19.1)
Requirement already satisfied: websocket-client<0.41,>=0.40.0 in /opt/octoprint/venv/lib/python2.7/site-packages/websocket_client-0.40.0-py2.7.egg (from OctoPrint->Bed-Visualizer==0.1.6) (0.40.0)
[... cut ...]
Requirement already satisfied: blinker in /opt/octoprint/venv/lib/python2.7/site-packages/blinker-1.4-py2.7.egg (from Flask-Principal<0.4,>=0.3.5->OctoPrint->Bed-Visualizer==0.1.6) (1.4)
Requirement already satisfied: pytz>=0a in /opt/octoprint/venv/lib/python2.7/site-packages/pytz-2018.5-py2.7.egg (from Babel>=1.0->Flask-Babel<0.10,>=0.9->OctoPrint->Bed-Visualizer==0.1.6) (2018.5)
Building wheels for collected packages: Bed-Visualizer, numpy
  Running setup.py bdist_wheel for Bed-Visualizer ... done
  Stored in directory: /tmp/pip-ephem-wheel-cache-ettyCf/wheels/77/77/fd/91c6afc2643e95d43d1fcf7ab4af9cd240dc9c42f72fe246f5
  Running setup.py bdist_wheel for numpy ... -
```

saikek, 2018-09-09:
pip 9.0.1 from /usr/lib/python2.7/dist-packages (python 2.7)

jneilliii, 2018-09-09:
Ok, so as others have posted, it looks like this is hanging on the numpy install. I would try manually installing numpy first and then the plugin, or choose a version of bed visualizer prior to 0.1.3 and install using that zip url in plugin manager and don't upgrade.

shafr, 2018-09-10:
Is there a way of Verbose install to see what's going on inside while it's inside numpy phase ?

jneilliii, 2018-09-10:
Not that I know of in octoprint and I'm not really a python expert so don't know much about pip options.

WarDrake, 2018-09-15:
/home/pi/oprint/bin/python2 -m pip install numpy 
Runnig this in the console fixes the issue, bed leveling can be easily installed after that

jangermann, 2018-10-04:
Hi. Also using a PI and have the same problem:

Installing plugin "Bed Level Visualizer" from https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip...
/home/pi/oprint/bin/python2 -m pip install https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip --no-cache-dir
Collecting https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip
Downloading https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from Bed-Visualizer==0.1.6)
Collecting numpy (from Bed-Visualizer==0.1.6)
Downloading https://files.pythonhosted.org/packages/45/ba/2a781ebbb0cd7962cc1d12a6b65bd4eff57ffda449fdbbae4726dc05fbc3/numpy-1.15.2.zip (4.5MB)
Requirement already satisfied: flask<0.11,>=0.10.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: Jinja2<2.9,>=2.8.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: werkzeug<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: tornado==4.5.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: PyYAML<3.11,>=3.10 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: Flask-Login<0.3,>=0.2.11 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: Flask-Babel<0.10,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: markdown<2.7,>=2.6.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: pyserial<3.5,>=3.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: netaddr<0.8,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: watchdog<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: sarge==0.1.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: netifaces<0.11,>=0.10.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: pylru<1.1,>=1.0.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: rsa<3.3,>=3.2.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: pkginfo<1.5,>=1.4.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: requests<3,>=2.18.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: semantic_version<2.5,>=2.4.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: psutil<5.5,>=5.4.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: Click<6.3,>=6.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: chainmap<1.1,>=1.0.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: future<0.16,>=0.15.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: scandir<1.4,>=1.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: websocket-client<0.41,>=0.40.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: python-dateutil<2.7,>=2.6.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: wrapt<1.11,>=1.10.11 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: futures<3.2,>=3.1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: emoji<0.5,>=0.4.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: monotonic<1.4,>=1.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->Bed-Visualizer==0.1.6)
[... cut ...]
Requirement already satisfied: six in /home/pi/oprint/lib/python2.7/site-packages (from websocket-client<0.41,>=0.40.0->OctoPrint->Bed-Visualizer==0.1.6)
Requirement already satisfied: pytz>=0a in /home/pi/oprint/lib/python2.7/site-packages (from Babel>=1.0->Flask-Babel<0.10,>=0.9->OctoPrint->Bed-Visualizer==0.1.6)
Installing collected packages: numpy, Bed-Visualizer
Running setup.py install for numpy: started
Running setup.py install for numpy: still running...
Running setup.py install for numpy: still running...

jneilliii, 2018-10-04:
You just have to wait, the numpy install takes forever. Some have reported up to an hour. Some have stated they had to manually install numpy first.

jangermann, 2018-10-04:
> You just have to wait, the numpy install takes forever. Some have reported up to an hour. Some have stated they had to manually install numpy first.

Thanks.
Actually octoprint did installed it. It was there 5 minutes AFTER an unknown error. 
I got it installed. Thanks.

ChrisWLeon, 2018-10-04:
my original problem remains.

i have uninstalled. reinstalled, installed earlier versions. but still no success. if i start in safe mode you can see the install was successful

jneilliii, 2018-10-04:
@ChrisWLeon, did you try what @WarDrake suggested?

> /home/pi/oprint/bin/python2 -m pip install numpy
> Runnig this in the console fixes the issue, bed leveling can be easily installed after that

If you run the following command what is returned?

```
cd ~/oprint/bin/
./pip freeze
```

ChrisWLeon, 2018-10-05:
Yes i tried all the suggestions including @warDrake ie

pi@octopi:~ $ /home/pi/oprint/bin/python2 -m pip install numpy
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: numpy in ./oprint/lib/python2.7/site-packages (1.15.1)

I manually uninstalled so listing installed plugins does not list bed visualiser ie

~/oprint/bin $ ./pip freeze
argh==0.26.2
awesome-slugify==1.6.5
Babel==2.5.3
backoff==1.6.0
backports-abc==0.5
backports.ssl-match-hostname==3.5.0.1
Bed-Leveling-Wizard==0.1.7
blinker==1.4
certifi==2018.4.16
chainmap==1.0.2
chardet==3.0.4
click==6.2
contextlib2==0.5.5
emoji==0.4.5
feedparser==5.2.1
Flask==0.10.1
Flask-Assets==0.10
Flask-Babel==0.9
Flask-Login==0.2.11
Flask-Principal==0.3.5
frozendict==1.2
future==0.15.2
futures==3.1.1
idna==2.6
itsdangerous==0.24
Jinja2==2.8.1
Markdown==2.6.11
MarkupSafe==1.0
monotonic==1.3
netaddr==0.7.19
netifaces==0.10.7
numpy==1.15.1
Octolapse==0.3.4rc1.dev3
OctoPrint==1.3.9
OctoPrint-Anywhere==0.8.3
OctoPrint-Autocalibration==0.1
pathtools==0.1.2
Pillow==5.2.0
pkg-resources==0.0.0
pkginfo==1.4.2
psutil==5.4.7
pyasn1==0.4.2
pybonjour==1.1.1
pylru==1.0.9
pyserial==3.4
python-dateutil==2.6.1
pytz==2018.4
PyYAML==3.10
ratelimit==1.4.1
raven==6.9.0
regex==2018.2.21
requests==2.18.4
rsa==3.2.3
sarge==0.1.4
scandir==1.3
semantic-version==2.4.2
singledispatch==3.4.0.3
six==1.11.0
sockjs-tornado==1.0.3
speaklater==1.3
tornado==4.5.3
Unidecode==0.4.21
urllib3==1.22
watchdog==0.8.3
webassets==0.12.1
websocket-client==0.40.0
Werkzeug==0.8.3
wrapt==1.10.11

so, I reinstalled it

pi@octopi:~/oprint/bin $ ./pip freeze
argh==0.26.2
awesome-slugify==1.6.5
Babel==2.5.3
backoff==1.6.0
backports-abc==0.5
backports.ssl-match-hostname==3.5.0.1
Bed-Leveling-Wizard==0.1.7
Bed-Visualizer==0.1.6
blinker==1.4
certifi==2018.4.16
chainmap==1.0.2
chardet==3.0.4
click==6.2
contextlib2==0.5.5
emoji==0.4.5
feedparser==5.2.1
Flask==0.10.1
Flask-Assets==0.10
Flask-Babel==0.9
Flask-Login==0.2.11
Flask-Principal==0.3.5
frozendict==1.2
future==0.15.2
futures==3.1.1
idna==2.6
itsdangerous==0.24
Jinja2==2.8.1
Markdown==2.6.11
MarkupSafe==1.0
monotonic==1.3
netaddr==0.7.19
netifaces==0.10.7
numpy==1.15.1
Octolapse==0.3.4rc1.dev3
OctoPrint==1.3.9
OctoPrint-Anywhere==0.8.5
OctoPrint-Autocalibration==0.1
pathtools==0.1.2
Pillow==5.2.0
pkg-resources==0.0.0
pkginfo==1.4.2
psutil==5.4.7
pyasn1==0.4.2
pybonjour==1.1.1
pylru==1.0.9
pyserial==3.4
python-dateutil==2.6.1
pytz==2018.4
PyYAML==3.10
ratelimit==1.4.1
raven==6.9.0
regex==2018.2.21
requests==2.18.4
rsa==3.2.3
sarge==0.1.4
scandir==1.3
semantic-version==2.4.2
singledispatch==3.4.0.3
six==1.11.0
sockjs-tornado==1.0.3
speaklater==1.3
tornado==4.5.3
Unidecode==0.4.21
urllib3==1.22
watchdog==0.8.3
webassets==0.12.1
websocket-client==0.40.0
Werkzeug==0.8.3
wrapt==1.10.11

still will not load the plugin, it is installed but will only appear in list of installed plugins if you start OctiPrint in safe mode

not sure what else to try… i have also installed earlier versions

jneilliii, 2018-10-05:
And there's no errors in your octoprint.log or browser's console indicating an issue when running in normal mode?

Prunar, 2018-10-10:
Numpy seems to be compiling itself but fails hard. Even after stopping Octoprint for some extra free memory it doesn't want to compile.

Complete output: https://pastebin.com/ZsxVGDdz

jneilliii, 2018-10-10:
Sorry @Prunar, but I can't do anything about that. Unless you need to support one of the firmwares added since 0.1.2, you can use the url below in plugin manager and make sure not to upgrade.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.2.zip

Prunar, 2018-10-10:
Yes thanks, 0.1.2 seems to work fine 👍

Dutchy-79, 2018-10-20:
Just tried this on a Pi 0 and keeps failing on numpy error after several minutes.

jneilliii, 2018-10-20:
@Dutchy-79, you mean the install fails? Yes read through this thread for possible options.

mcc05, 2018-10-21:
Im have the same issue, in the virtual env, I try a pip install numpy -v, and it freezes at arm-linux-gnueabihf-gcc: build/src.linux-armv7l-2.7/numpy/core/src/npysort/quicksort.c

This just maxes out the CPU and hangs

jneilliii, 2018-10-21:
Have any of you tried plugging in your device to ethernet vs using wifi?

wapowell, 2018-11-04:
Hi.

I'm having the exact same symptoms as ChrisWLeon.  I made sure nampy is installed.  I install bed visualizer via the octoprint web front end and all seems successful.  I ssh to the pi and verify it's there using pip freeze.  But in Octoprint the plugin is only listed as installed when in safe mode.  In regular mode it doesn't show up.

Any ideas?

jneilliii, 2018-11-05:
What version is your octoprint? There was one version that blacklisted the plugin because it would kill the interface altogether. Are there any errors in your logs?

wapowell, 2018-11-05:
> 
> 
> What version is your octoprint? There was one version that blacklisted the plugin because it would kill the interface altogether. Are there any errors in your logs?

I'm on 1.3.9.  I haven't checked error logs yet but I can do that when I get home.  Was there ever a resolution for the other that posted in this thread?

ChrisWLeon, 2018-11-08:
> I'm not 100% sure if they ever resolved the issue or not.

I was never able to resolve the problem. Ended up reformatting and reinstalling octoprint. It did install but now it will not save the data

Dr. Christopher Leon BVSc (Hons), GradDipQMgtHlth. RN.

closed 2018-11-29 by stale[bot]

nordurljosahvida, 2020-08-14:
> /home/pi/oprint/bin/python2 -m pip install numpy
> Runnig this in the console fixes the issue, bed leveling can be easily installed after that

This took hours for me to complete even after setting up swap space but it ended up working. Thanks!

jneilliii, 2020-08-14:
Once octoprint has migrated to python 3 completely this will get much better thanks to piwheels.

locki-cz, 2021-01-31:
CCompilerOpt._cache_write[796] : write cache to path -> /tmp/pip-install-eicidf7l/numpy/build/temp.linux-armv7l-3.8/ccompiler_opt_cache_clib.py
  error: [Errno 28] No space left on device
  ----------------------------------------
  ERROR: Failed building wheel for numpy
Failed to build numpy
ERROR: Could not build wheels for numpy which use PEP 517 and cannot be installed directly

env.os.id: linux
env.os.platform: linux
env.python.pip: 20.3.3
env.python.version: 3.8.5
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.3

root@orangepione:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            188M     0  188M   0% /dev
tmpfs            50M  4.4M   45M   9% /run
/dev/mmcblk0p1   30G  2.5G   27G   9% /
tmpfs           246M     0  246M   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           246M     0  246M   0% /sys/fs/cgroup
tmpfs           246M  524K  246M   1% /tmp
/dev/zram1       49M   20M   25M  45% /var/log
tmpfs            50M     0   50M   0% /run/user/0

Device: orange pi one

how much space it need?