jneilliii/OctoPrint-BedLevelVisualizer #82: Add support for Repetier Firmware G33

gztproject, 2018-09-06:
Hi,

I'm using repetier-firmware where G33 L0 outputs the correction matrix in format
`Send: N20 G33 L0*119
Recv: ok 20
Recv: G33 X32.50 Y15.00 Z-0.076
...`

and G33  (ie. actual probing) outputs in format
`Recv: Z-probe:20.199 X:32.50 Y:241.91`

Any chance you could include support for either one? I can try and do it myself when I can find some time.

jneilliii, 2018-09-06:
The G33 L0 output I think would be able to be accounted for with some minor tweaks as that looks very similar to the old marlin and smoothieware formats that i just recently updated for. By chance have you tried the plugin yet?

jneilliii, 2018-09-06:
And can you take a screenshot of the output please, looks like the returns got lost in your quote. Or better yet, surround the output with three back ticks like this `````

gztproject, 2018-09-06:
Just submitted a pull request :)

jneilliii, 2018-09-06:
That's awesome.  I just merged and will update the other stuff necessary to make a full release version shortly.