jneilliii/OctoPrint-BedLevelVisualizer #85: 0.1.6 intallation does not ends.

Thisismydigitalself, 2018-09-07:
OctoPrint ver 1.3.9

Requirement already satisfied: pytz>=0a in /home/pi/oprint/lib/python2.7/site-packages (from Babel>=1.0->Flask-Babel<0.10,>=0.9->OctoPrint->Bed-Visualizer==0.1.6) (2018.4)Installing collected packages: numpy, Bed-VisualizerRunning setup.py install for numpy: startedRunning setup.py install for numpy: 
still running...Running setup.py install for numpy: 
still running...Running setup.py install for numpy: 
still running...Running setup.py install for numpy: 
still running...Running setup.py install for numpy: 
still running...Running setup.py install for numpy: 
still running...Running setup.py install for numpy: 
still running...

Please advice steps for fixing this.
TY.

Thisismydigitalself, 2018-09-07:
UPDATE: Installation finished after 5 min or so.
[image]

jneilliii, 2018-09-07:
Yeah, the numby requirement incorporated since version 0.1.3 introduced this slowness.  As documented in the [Known Issues](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer#known-issues). Some have reported it takes over 30 minutes depending on the Raspberry model.

closed 2018-09-07 by jneilliii

Thisismydigitalself, 2018-09-08:
If possible let the installation prompt the user things are progressing. for a noob like myself and I guess others like me it was not clear what goes on under the hood and no way to cancel.