jneilliii/OctoPrint-BedLevelVisualizer #87: Tevo Tornado

debil1, 2018-09-22:
Hi,

I installed this plugin but don't know how to use it.

I manually inserted commands into Terminal:  G29 T and received back:

Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 -0.015 +0.011 +0.048 +0.017 -0.054
Recv:  1 -0.040 -0.066 +0.010 -0.039 -0.043
Recv:  2 -0.008 +0.045 +0.051 +0.034 -0.057
Recv:  3 +0.026 +0.033 +0.050 -0.005 -0.033
Recv:  4 +0.074 +0.051 +0.087 +0.053 -0.096

But where can I see visualization of that ?
At Bed visualizer I have only to insert:  " GCODE Commands" no pictures. At settings I have same + stored data.

debil1, 2018-09-22:
Chrome, I just tried to put into terminal

G28
@BEDLEVELVISUALIZER
G29 T

debil1, 2018-09-22:
Or is this not correct method to try it ?

jneilliii, 2018-09-22:
That should work. Are you saying the bed visualizer tab isn't showing anything at all? The tab could be hidden in the far right pull down, maybe that's what you're missing.

MimbaMonkeyHouse, 2018-09-26:
This works fine for me on a Tevo Tornado with Octoprint 1.3.9:
G28
@BEDLEVELVISUALIZER
G29 T

closed 2018-10-17 by stale[bot]