jneilliii/OctoPrint-BedLevelVisualizer #89: ABL_BININEAR_SUBDIVISION causes unexpected graphing output

chainsol, 2018-10-03:
### Issue
Yesterday on the Freenode #reprap channel, someone sent a screenshot from the Octoprint bed level visualizer, with a bit of strange behavior:
[image]

We asked for a printout of the G29 associated with this, and the result is as follows:
```
Recv:       0      1      2      3      4      5      6      7      8
Recv:  0 -0.043 -0.121 -0.234 -0.344 -0.394 -0.285 -0.269 -0.166 -0.086
Recv:  1 -0.086 -0.120 -0.338 -0.352 -0.366 -0.335 -0.246 -0.072 -0.017
Recv:  2 -0.146 -0.300 -0.329 -0.323 -0.323 -0.252 -0.192 -0.133 +0.100
Recv:  3 -0.213 -0.321 -0.339 -0.316 -0.274 -0.203 -0.135 -0.080 +0.185
Recv:  4 -0.347 -0.365 -0.357 -0.302 -0.202 -0.144 -0.107 -0.028 +0.020
Recv:  5 -0.241 -0.420 -0.380 -0.314 -0.168 -0.097 -0.051 +0.010 +0.166
Recv:  6 -0.384 -0.454 -0.377 -0.243 -0.142 -0.047 +0.043 +0.075 +0.171
Recv:  7 -0.526 -0.435 -0.409 -0.249 -0.139 -0.024 +0.087 +0.206 +0.212
Recv:  8 -0.660 -0.520 -0.322 -0.145 -0.119 +0.053 +0.125 +0.209 +0.269
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5        6        7        8        9       10       11       12       13       14       15       16       17       18       19       20       21       22       23       24
Recv:  0 -0.04315 -0.06786 -0.09258 -0.12118 -0.15641 -0.19553 -0.23425 -0.27321 -0.31177 -0.34372 -0.37066 -0.39098 -0.39365 -0.36566 -0.32002 -0.28469 -0.27562 -0.27686 -0.26852 -0.23992 -0.20175 -0.16606 -0.13772 -0.11187 -0.08602
Recv:  1 -0.05698 -0.07230 -0.08763 -0.11422 -0.16269 -0.22242 -0.27285 -0.30537 -0.32858 -0.34765 -0.36723 -0.38267 -0.38517 -0.36622 -0.33434 -0.30632 -0.29233 -0.28220 -0.26216 -0.22176 -0.17146 -0.12886 -0.10239 -0.08363 -0.06487
Recv:  2 -0.07080 -0.07674 -0.08267 -0.10725 -0.16896 -0.24932 -0.31144 -0.33753 -0.34540 -0.35159 -0.36381 -0.37435 -0.37669 -0.36678 -0.34867 -0.32795 -0.30904 -0.28753 -0.25580 -0.20360 -0.14117 -0.09165 -0.06707 -0.05539 -0.04371
Recv:  3 -0.08647 -0.09096 -0.09544 -0.12031 -0.18669 -0.27345 -0.33762 -0.35727 -0.35433 -0.35150 -0.35813 -0.36488 -0.36645 -0.36148 -0.35134 -0.33483 -0.31260 -0.28400 -0.24593 -0.18975 -0.12410 -0.07174 -0.04470 -0.03095 -0.01720
Recv:  4 -0.10496 -0.12373 -0.14249 -0.17260 -0.22786 -0.29448 -0.34239 -0.35507 -0.34906 -0.34394 -0.34827 -0.35349 -0.35348 -0.34622 -0.33373 -0.31590 -0.29313 -0.26502 -0.23028 -0.18472 -0.13253 -0.08490 -0.04708 -0.01382 +0.01944
Recv:  5 -0.12530 -0.16628 -0.20725 -0.24492 -0.28049 -0.31274 -0.33475 -0.34045 -0.33591 -0.33235 -0.33616 -0.34095 -0.33874 -0.32510 -0.30447 -0.28222 -0.26051 -0.23718 -0.21111 -0.18402 -0.15418 -0.11535 -0.06242 -0.00049 +0.06144
Recv:  6 -0.14640 -0.20209 -0.25778 -0.29971 -0.32002 -0.32656 -0.32916 -0.32954 -0.32597 -0.32310 -0.32537 -0.32833 -0.32336 -0.30456 -0.27784 -0.25227 -0.23123 -0.21134 -0.19168 -0.17845 -0.16544 -0.13305 -0.06831 +0.01581 +0.09992
Recv:  7 -0.16570 -0.22154 -0.27738 -0.31777 -0.33334 -0.33347 -0.33086 -0.32882 -0.32404 -0.31933 -0.31800 -0.31673 -0.30839 -0.28778 -0.26011 -0.23383 -0.21175 -0.19107 -0.17192 -0.16281 -0.15522 -0.12368 -0.05174 +0.04415 +0.14004
Recv:  8 -0.18576 -0.23425 -0.28273 -0.31830 -0.33357 -0.33592 -0.33461 -0.33180 -0.32532 -0.31790 -0.31195 -0.30506 -0.29277 -0.27158 -0.24499 -0.21912 -0.19561 -0.17281 -0.15189 -0.14230 -0.13460 -0.10157 -0.02572 +0.07546 +0.17664
Recv:  9 -0.21327 -0.25263 -0.29199 -0.32133 -0.33552 -0.33968 -0.33925 -0.33515 -0.32646 -0.31578 -0.30444 -0.29113 -0.27448 -0.25285 -0.22789 -0.20326 -0.17983 -0.15674 -0.13507 -0.12341 -0.11317 -0.07965 -0.00685 +0.08924 +0.18533
Recv: 10 -0.26182 -0.28817 -0.31452 -0.33401 -0.34371 -0.34655 -0.34449 -0.33752 -0.32564 -0.31080 -0.29291 -0.27207 -0.25051 -0.22846 -0.20569 -0.18378 -0.16367 -0.14442 -0.12473 -0.10907 -0.09297 -0.06178 -0.00556 +0.06576 +0.13708
Recv: 11 -0.31783 -0.32939 -0.34095 -0.34918 -0.35362 -0.35473 -0.35062 -0.34026 -0.32468 -0.30513 -0.27993 -0.25076 -0.22387 -0.20155 -0.18150 -0.16314 -0.14787 -0.13429 -0.11760 -0.09634 -0.07196 -0.04409 -0.01142 +0.02475 +0.06092
Recv: 12 -0.34715 -0.35424 -0.36132 -0.36546 -0.36645 -0.36449 -0.35732 -0.34413 -0.32572 -0.30225 -0.27047 -0.23363 -0.20157 -0.17838 -0.15997 -0.14391 -0.13174 -0.12191 -0.10733 -0.08287 -0.05365 -0.02797 -0.00978 +0.00487 +0.01951
Recv: 13 -0.32032 -0.34438 -0.36843 -0.38348 -0.38503 -0.37759 -0.36564 -0.35149 -0.33284 -0.30731 -0.26957 -0.22495 -0.18707 -0.16183 -0.14333 -0.12753 -0.11549 -0.10615 -0.09220 -0.06928 -0.04176 -0.01543 +0.00737 +0.02897 +0.05057
Recv: 14 -0.26682 -0.31815 -0.36948 -0.40260 -0.40654 -0.39228 -0.37454 -0.35998 -0.34195 -0.31516 -0.27218 -0.22044 -0.17691 -0.14903 -0.12936 -0.11255 -0.09891 -0.08813 -0.07392 -0.05495 -0.03256 -0.00446 +0.03202 +0.07420 +0.11638
Recv: 15 -0.24097 -0.30854 -0.37612 -0.41953 -0.42366 -0.40361 -0.38050 -0.36326 -0.34295 -0.31381 -0.26819 -0.21373 -0.16768 -0.13766 -0.11604 -0.09717 -0.08062 -0.06682 -0.05133 -0.03567 -0.01834 +0.00964 +0.05480 +0.11060 +0.16641
Recv: 16 -0.27010 -0.33295 -0.39580 -0.43437 -0.43367 -0.40870 -0.38011 -0.35540 -0.32709 -0.29334 -0.24962 -0.20048 -0.15772 -0.12672 -0.10211 -0.07963 -0.05815 -0.03881 -0.02070 -0.00775 +0.00396 +0.02706 +0.06952 +0.12337 +0.17721
Recv: 17 -0.32689 -0.37398 -0.42107 -0.44701 -0.43929 -0.41042 -0.37679 -0.34233 -0.30310 -0.26367 -0.22445 -0.18502 -0.14870 -0.11721 -0.08882 -0.06168 -0.03397 -0.00750 +0.01424 +0.02513 +0.03129 +0.04759 +0.08238 +0.12731 +0.17224
Recv: 18 -0.38364 -0.41246 -0.44127 -0.45380 -0.44132 -0.41255 -0.37730 -0.33544 -0.28711 -0.24258 -0.20638 -0.17399 -0.14215 -0.10979 -0.07798 -0.04675 -0.01413 +0.01793 +0.04350 +0.05608 +0.06217 +0.07538 +0.10260 +0.13694 +0.17128
Recv: 19 -0.43141 -0.43960 -0.44779 -0.44771 -0.43711 -0.41824 -0.38954 -0.34560 -0.29182 -0.24290 -0.20531 -0.17257 -0.14004 -0.10623 -0.07261 -0.03899 -0.00367 +0.03165 +0.06206 +0.08408 +0.10120 +0.11896 +0.13945 +0.16058 +0.18172
Recv: 20 -0.47915 -0.46419 -0.44924 -0.43575 -0.42931 -0.42433 -0.40562 -0.36194 -0.30451 -0.25180 -0.21134 -0.17559 -0.14042 -0.10476 -0.06968 -0.03426 +0.00245 +0.03949 +0.07494 +0.11016 +0.14378 +0.16979 +0.18366 +0.18992 +0.19618
Recv: 21 -0.52597 -0.49338 -0.46079 -0.43531 -0.42662 -0.42502 -0.40862 -0.36328 -0.30313 -0.24858 -0.20851 -0.17405 -0.13895 -0.10075 -0.06192 -0.02359 +0.01337 +0.04983 +0.08721 +0.13050 +0.17472 +0.20637 +0.21675 +0.21456 +0.21236
Recv: 22 -0.57128 -0.53109 -0.49089 -0.45601 -0.43364 -0.41658 -0.38854 -0.33757 -0.27563 -0.22230 -0.18857 -0.16345 -0.13358 -0.09182 -0.04530 -0.00202 +0.03374 +0.06626 +0.10034 +0.14192 +0.18507 +0.21677 +0.22885 +0.22947 +0.23010
Recv: 23 -0.61567 -0.57339 -0.53111 -0.48822 -0.44578 -0.40273 -0.35539 -0.29687 -0.23405 -0.18390 -0.15977 -0.14831 -0.12637 -0.08035 -0.02385 +0.02549 +0.05892 +0.08518 +0.11285 +0.14761 +0.18378 +0.21290 +0.22982 +0.23969 +0.24956
Recv: 24 -0.66007 -0.61569 -0.57132 -0.52044 -0.45791 -0.38889 -0.32224 -0.25616 -0.19247 -0.14549 -0.13097 -0.13317 -0.11916 -0.06888 -0.00240 +0.05299 +0.08409 +0.10409 +0.12536 +0.15329 +0.18248 +0.20904 +0.23079 +0.24991 +0.26903
```

It appears that when given the subdivided grid, this plugin gets a little confused!
### Resolution
If you turn off ABL_BININEAR_SUBDIVISION in Marlin, this issue goes away - the mesh looks normal.
[image]

jneilliii, 2018-10-03:
Thanks for the detailed information.  When you have the subdivision enabled and use the store mesh option in the plugin's settings what is the stored data look like?  Does it match the info you show above? The only reason I ask is because there are numbers in your response above that looks like they are way off from the norm and so that may be what's causing the unexpected results here.

[image]

chainsol, 2018-10-03:
Hmmm, I'll have to have the original user who reported this test that. Is there a way to see previously stored results?
That would mean that there's a bug in the subdivision, wouldn't it?

jneilliii, 2018-10-03:
Yeah, that's what i'm thinking, issue with the subdivision. The results are saved when you run the configured commands, no historical data just the last run.

chainsol, 2018-10-03:
OK, the original person with the problem is having serious issues with G29 in Marlin, so they're still working on testing it out again. I have a hunch that you're right - it's a subdivision issue.

tfischer4765, 2018-10-04:
This is the aforementioned original person speaking. I'm sorry to report that I've still not worked out my differences with Marlin to a point where I can once more issue a G29 without the printer doing its level best (pun intended) to self-destruct. The last result was quite spectacular, with one drive sled ending up BELOW the delta effector. I will, however, continue my quest to work this out, and provided my printer doesn't succeed in its suicidal endeavors before I succeed in restoring it to working order, I should be getting back to you in a few days.

jneilliii, 2018-10-04:
No problem @tfischer4765.  I was having issues with the latest Marlin 2.0.x bugfix branch with autoleveling where my probe wasn't triggering properly and my nozzle would crash into the bed and ended up reverting to the official stable Marlin 1.1.9 release.

closed 2018-10-25 by stale[bot]