jneilliii/OctoPrint-BedLevelVisualizer #90: Support for Smoothieware Cartesian Grid Leveling Mesh

CapnBry, 2018-10-07:
Smoothieware's Cartesian grid leveling system data doesn't trigger parsing properly in the BedLevelVisualizer plugin. To display the data a `M375.1` must be sent and the data _almost_ works.
```
  270.0000|    0.4713     0.4325     0.4200     0.4100     0.3838     0.3363     0.2763     0.2088     0.1213
  236.2500|    0.4038     0.3750     0.3588     0.3613     0.3512     0.3188     0.2700     0.1963     0.1163
  202.5000|    0.3325     0.2988     0.2963     0.3025     0.2950     0.2738     0.2363     0.1838     0.1163
  168.7500|    0.2613     0.2313     0.2387     0.2525     0.2538     0.2463     0.2213     0.1762     0.1113
  135.0000|    0.2138     0.1888     0.1950     0.2225     0.2325     0.2350     0.2238     0.1900     0.1525
  101.2500|    0.1475     0.1488     0.1788     0.2225     0.2425     0.2650     0.2600     0.2387     0.2250
   67.5000|    0.1012     0.1125     0.1513     0.1963     0.2350     0.2575     0.2600     0.2650     0.2650
   33.7500|    0.0338     0.0612     0.1175     0.1637     0.1963     0.2350     0.2487     0.2687     0.0663
    0.0000|   -0.0013     0.0250     0.0838     0.1463     0.1600     0.2000     0.2275     0.2438     0.2550
           -----+----------+----------+----------+----------+----------+----------+----------+----------+-----
           0.0000 31.2500 62.5000 93.7500 125.0000 156.2500 187.5000 218.7500 250.0000
Probe completed
ok
```

It is just the first column throwing off the detection regex. If the regex [in __init__.py](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/octoprint_bedlevelvisualizer/__init__.py#L69) is changed to include a matching rule, it works great, along with the existing "Remove Row Labels" and "Flip Y Axis" options. The match is `([\d.]+\|\s+)` so the full regex needed is
```
... and re.match(r"^((G33.+)|(Bed.+)|(\d+\s)|(\|\s+)|([\d.]+\|\s+)|(\[?\s?\+?\-?\d?\.\d+\]?\s*\,?)|(\s?\.\s*)|(NAN\,?))+$", line.strip()):
```

Tested on [latest Smoothieware](https://github.com/Smoothieware/Smoothieware/tree/edge/FirmwareBin) and OctoPrint 1.3.9, although I do not have any Marlin or Repetier firmwares to regression test.

jneilliii, 2018-10-07:
Hey @CapnBry, is this a new format that Smoothieware is using now? Based on [this](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/22#issuecomment-384697647) post the format is a little different.  I'm going through some tests now, and I think this does actually break some of the other formats.

BTW, it's your fault I got into 3D printing/OctoPrint in the first place because of the amazing HeaterMeter project, thanks for the work you do on that.

CapnBry, 2018-10-07:
Oh awesome! Well see how great that is? This morning I was like "I wonder if someone made an OctoPrint plugin to display the bed leveling, that would be super swell" and there was! It took numpy 5x longer to compile and install than it did to get it working, so thanks for making this.

The grid output in smoothieware isn't standard across the different modules. The 3-point strategy, the delta strategy, the delta grid strategy, and the cartesian grid strategy all output information in different formats. Their Gxx and Mxx gcode commands don't even use the same numbers, like M374 saves the delta grid but M375 saves the cartesian grid. G32 does a delta grid probe but G29 does a cartesian grid probe. I think delta grid and cartesian grid are the only two that would be relevant to this plugin though, which has the output you linked, and this one is the cartesian grid output.

Is there a better regex that would match on the "  270.0000|" part, or a way it can be ignored? It is very similar to the `(\|\s+)` capture apart from starting with a number. It would be great if I could get this fixed upstream so that all the Smoothieware grids come out looking like the Marlin ones but I wouldn't hold my breath for that. Let me know if there's something I can do to help you resolve this.

CapnBry, 2018-10-08:
Welp I am an idiot. I looked into the Smoothieware source code to make it output the proper format... and found it already has this as an option.
```
leveling-strategy.rectangular-grid.human_readable false
```
Setting that option doesn't put out the leading numbers or ruler bar on the bottom, and actually prints them in the proper order so "Flip Y" isn't even needed. Here's the output for posterity
```
Send: M375.1

0.0000 -0.0038 0.0275 0.0600 0.0538 0.0612 0.0575 0.0488 0.0300
0.0263 0.0238 0.0375 0.0638 0.0738 0.0812 0.0700 0.0550 0.0213
0.0638 0.0375 0.0500 0.0675 0.0750 0.0712 0.0450 0.0238 -0.0113
0.0800 0.0475 0.0500 0.0587 0.0563 0.0438 0.0125 -0.0325 -0.0700
0.1100 0.0625 0.0350 0.0325 0.0200 -0.0100 -0.0550 -0.1163 -0.1725
0.1275 0.0725 0.0512 0.0350 0.0125 -0.0213 -0.0838 -0.1575 -0.2437
0.1600 0.1100 0.0750 0.0500 0.0238 -0.0275 -0.0938 -0.1750 -0.2713
0.2125 0.1500 0.1137 0.0888 0.0438 -0.0087 -0.0863 -0.1925 -0.2938
0.2525 0.1850 0.1363 0.1037 0.0463 -0.0162 -0.1050 -0.2088 -0.3187
ok
```

Looks like there's no changes needed at all for this to work with smoothieware apart from turning off human_readable in the firmware config. Sorry about the noise!

closed 2018-10-08 by CapnBry