jneilliii/OctoPrint-BedLevelVisualizer #91: I'm getting what looks like a Javascript error when running the plugin on an MK3 (TypeError: undefined is not an object (evaluating 't._fullLayout._defs.selectAll'))

edgarschmidt, 2018-10-13:
I'm getting the following error while upon running the plugin on my Prusa MK3:

`**TypeError:** undefined is not an object (evaluating 't._fullLayout._defs.selectAll')**`

Can anyone help me? These are the commands I entered for this printer:

`G80
@BEDLEVELVISUALIZER
G81`

jneilliii, 2018-10-13:
Are the commands each on their own line like this?

```
G80
@BEDLEVELVISUALIZER
G81
```

edgarschmidt, 2018-10-13:
Yes

- Edgar

edgarschmidt, 2018-10-13:
Alright, the browser and the error log didn't show anything unusual (I'm running this on my MBP 13 Tocuhbar), however I ran updates again and rebooted Octopi and now everything works! I have no idea why I was seeing that error before.

edgarschmidt, 2018-10-13:
Will this plugin report back changes made through the manual tuning option in the MK3 controller?

jneilliii, 2018-10-13:
I'm not really sure how the prusa firmware works, but assuming it stores the leveling data in eeprom the command G81 would just report the current data so making the plugin's settings the following it should show. 

```
@BEDLEVELVISUALIZER
G81
```

closed 2018-11-03 by stale[bot]